vitormhenrique/OctoPrint-Enclosure #252: DS18B20 Temperature Warnings via MQTT

coolduke666, 2019-08-17:
**Is your feature request related to a problem? Please describe.**
Apart from IFTT it would be great to have a MQTT implementation for serveral things. upmost importpant would be a trigger for a temperatur alarm via MQTT which should turn off the printer (via smarthome socket plug).

**Describe the solution you'd like**
it would be great to have a cooperation between the temp.values from your plugin and the already existing MQTT plugin. 

With the enclsoure plugin I cant read the temps from my DS18B20. with the MQTT I could send the temp.data form the bed and extruder. 

But as a failfafe if the marlin boards goes crazy, it would be very handy to send the DS18B20 value via MQTT to whatever smarthome solution you have in place. 

**Describe alternatives you've considered**
none.
**Additional context**
your plugin is absolutely great but I think that feature can really make much more popular :)

coolduke666, 2019-08-25:
any Feedback?

vitormhenrique, 2019-08-25:
MQTT feature request is already there, I'm currently working on other projects and will implement this whenever I have the time, pull requests are always welcome, so if anyone wants to contribute this should not be that hard to implement. I'm gonna close this one because it's basically a duplicated issue as #40

closed 2019-08-25 by vitormhenrique

coolduke666, 2019-08-26:
thx... I missed that.

one thing is still missing: an API call for READING the Temps... currently only setting the temps are implemented: https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/API-Control

how to read the temperature on the DS18B20 via API-control?

thx