vitormhenrique/OctoPrint-Enclosure #254: Temperature Alarm doesn't work as it should

TechLifeTravel, 2019-08-30:
Relay doesn't switch OFF when temperature is reached threshold.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect relay to GPIO 12 (3.3v to VCC, 5V to JD-VCC and Ground to GND), DS18B20 sensor to GPIO 4.
Active Low parameter set.
2. Turn relay ON via web interface.
3. Create Output - Temperature Alarm
Set Temperature 31
Set Controlled IO Value LOW
4. Hold sensor in hand, when 31 is reached nothing happened
5. Set Controlled IO Value HIGH
6. Hold sensor in hand, when 31 is reached nothing happened
7. If I manually turn OFF the relay while the temperature is above the threshold, the relay switches itself ON again, regardless of the parameters Set Controlled IO Value (low or high)

**Expected behavior**
Relay should switch off when temperature is reached threshold

**Additional context**
My screenshots and log attached:
[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/3559990/octoprint.log)
[image]
[image]
[image]
[image]
[image]
[image]

TechLifeTravel, 2019-09-05:
Not a single answer in 6 days, is **this project is abandoned**? sad... This plugin could be very helpful :(

vitormhenrique, 2019-09-05:
it's not abandoned, it's a piece of free software that I spent months and months developing and maintaining, but i also have a life, full time job, vacations and other projects to work on... 
Giving that your github has ZERO projects and zero contributions what is **sad** is you thinking that I own you some sort of quick response or anything at all, source code is there, if you think you have an issue or bug and you don't feel like waiting, just fix it yourself.
This project IS very helpful, i have hundreds and hundreds of people messaging me, tanking me, asking questions and posting cool stuff that they build using it, some even giving me small donations, what have you done for the project? or any open source project for that matter?

TechLifeTravel, 2019-09-05:
Wow! You spent so much time on this aggressive answer, it would be better if you spent so much energy on the answer to the essence of the question. **I just asked if it was worth waiting for an answer.** I did not ask you to deal with this problem immediately. I respect your work and understand that this is a voluntary matter. **Sorry for this misunderstanding.** I hope you will have time to check my problem. If it is present at all, then this means that the anti-fire emergency shutdown settings do not work and may result in a fire.

TechLifeTravel, 2019-09-05:
**"what have you done for the project?"** I found a problem in the hope that the developer is interested in a correctly working application. I am not a programmer for this project, I am an end user. All I can do is search for bugs and suggest ideas. I am shocked by such a reaction to a simple question. Hundreds of grateful users will cease to be so if their house burns down, because you did not have the desire to fix a serious bug in the automation.

igorenot, 2019-09-06:
Hi guys,
I can confirm this issue!
I configured the sensors to turn off the printer in case of fire. I was sure that everything works. After reading this report, I decided to check, and conducted a test with a hot air gun. No, it does not work. Exactly as described in the issue. And if the printer was turned off, then when the specified temperature (fire) is reached, it will turn on !!! Given the seriousness of the detected problem, the developer should pay close attention to this. Thank you in advance!

gromeck, 2020-08-24:
G'day,
I forked and fixed it. See here:
https://github.com/gromeck/OctoPrint-Enclosure/commit/d74c302af7e8cb95453e8252bba81ac090b726cb#diff-c0b90a1bb8b923809f50ad50ff4ba80a

I would appreciate your feedback if it also works in your setup.

--gromeck

vitormhenrique, 2020-10-19:
merged @gromeck changes on master...

closed 2020-10-19 by vitormhenrique