vitormhenrique/OctoPrint-Enclosure #258: Outputs - Temperature Alarm

Tropper71, 2019-09-25:
Hello,
I can`t select any "Controlled IO"! I see the Pulldown Menu but no entry or selection!
Regards
Ralf

[image]

igorenot, 2019-09-25:
Do you have working outputs?

vitormhenrique, 2019-09-25:
You have to configure outputs first....

Tropper71, 2019-09-26:
How can I do this? Per PuTTy, Octoprint or config.txt and that is the syntax?
Thanks for your help.

Tropper71, 2019-09-27:
How can I do this? Per PuTTy, Octoprint or config.txt and that is the syntax?
For example GPIO25 defined as OUT but the sane result.
[image]

Thanks for your help.

vitormhenrique, 2019-09-27:
Read the documentation on GitHub, you need to setup output on the enclosure
plugin.

Tropper71, 2019-09-27:
" ...need to setup output on the enclosure..." Yes but how can I do this? I can`t find any solution.

vitormhenrique, 2019-09-27:
I’m on vacation, setting up the plugin is very self explanatory, you add
outputs, or inputs, The same way you tried to configure the input you do
for the output....
read the configuration page on the enclosure settings.

Tropper71, 2019-09-29:
Some things are so easy, sorry for the disturbance on your vacation! The problem is solved!
First of all I must generate a new output type **"Regular IO "** and then I can see and use this output in the menu **"Controlled IO"** of **"Temperature Alarm "**. OK now!

vitormhenrique, 2019-09-29:
No worries! Glad you figure it out!

closed 2019-10-21 by vitormhenrique