vitormhenrique/OctoPrint-Enclosure #262: No IFTTT activity on "Notify gpio actions"

oblivionratula, 2019-10-10:
I have "Notify gpio actions" checked and have my working Key set in the the Advanced Settings. GPIOs properly toggle when I click the buttons and auto startup and shutdown work, but there is no network traffic to IFTTT and setting the enclosure plugin log to DEBUG only shows the GPIO activity but no indication it's trying to communicate w/ IFTTT at all.

vitormhenrique, 2019-10-10:
Did you also setup a custom hook on the IFTTT  using the template that I
have documented (somewhere) ?

oblivionratula, 2019-10-10:
Yes, I did. As I said, there isn't even any network traffic going out towards IFTTT, which I can clearly see generated by the IFTTT plugin itself.

oblivionratula, 2019-10-10:
[image]
----------------------------------------
[image]