vitormhenrique/OctoPrint-Enclosure #265: Can't get DHT11 to work

joeynuggetz, 2019-10-18:
Logs complaining about unchecking "use sudo" but that is not enabled in settings. I have a 3 pin DHT11 plugged into BCM4 which worked fine on the dashboard plugin before they switched to using the sensor info from your plugin. Any advice? here's the log:

```
2019-10-18 03:53:56,176 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
2019-10-18 03:53:56,211 - octoprint.util.comm - INFO - Printer reports firmware name "Prusa-Firmware 3.8.0 based on Marlin"
2019-10-18 03:53:59,807 - octoprint.plugins.tracking - INFO - Sent tracking event printer_connected, payload: {u'printer_baudrate': 115200, u'printer_port': u'AUTO', 'firmware_name': 'Prusa-Firmware 3.8.0 based on Marlin'}
2019-10-18 03:54:05,041 - backoff - INFO - Backing off capture_mjpeg(...) for 13.8s (URLError: <urlopen error [Errno 111] Connection refused>)
2019-10-18 03:54:05,834 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2019-10-18 03:54:05,838 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('need more than 1 value to unpack',)
2019-10-18 03:54:15,963 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2019-10-18 03:54:15,967 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('need more than 1 value to unpack',)
2019-10-18 03:54:18,858 - backoff - INFO - Backing off capture_mjpeg(...) for 31.9s (URLError: <urlopen error [Errno 111] Connection refused>)
2019-10-18 03:54:26,092 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2019-10-18 03:54:26,096 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('need more than 1 value to unpack',)
2019-10-18 03:54:36,222 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2019-10-18 03:54:36,223 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('need more than 1 value to unpack',)
2019-10-18 03:54:46,345 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2019-10-18 03:54:46,349 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('need more than 1 value to unpack',)
2019-10-18 03:54:47,204 - tornado.access - WARNING - 404 GET /HNAP1/ (::ffff:10.0.0.43) 13.67ms
2019-10-18 03:54:47,220 - tornado.access - WARNING - 404 GET /HNAP1/ (::ffff:10.0.0.43) 7.13ms
2019-10-18 03:54:47,874 - tornado.access - WARNING - 404 POST /JNAP/ (::ffff:10.0.0.43) 664.78ms
2019-10-18 03:54:47,879 - tornado.access - WARNING - 404 POST /JNAP/ (::ffff:10.0.0.43) 654.83ms
2019-10-18 03:54:50,746 - backoff - INFO - Backing off capture_mjpeg(...) for 20.5s (URLError: <urlopen error [Errno 111] Connection refused>)
2019-10-18 03:54:56,477 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2019-10-18 03:54:56,481 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('need more than 1 value to unpack',)
2019-10-18 03:55:06,603 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2019-10-18 03:55:06,605 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('need more than 1 value to unpack',)
2019-10-18 03:55:11,235 - backoff - INFO - Backing off capture_mjpeg(...) for 22.4s (URLError: <urlopen error [Errno 111] Connection refused>)
2019-10-18 03:55:16,730 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2019-10-18 03:55:16,735 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('need more than 1 value to unpack',)

```

vitormhenrique, 2019-10-18:
Under the advanced settings, disable the use sudo.

joeynuggetz, 2019-10-18:
Thats not enabled.

vitormhenrique, 2019-10-21:
have you installed the required libraries?

Sarithz, 2020-03-28:
I'm having a similar issue with the DHT11.  All the command routines to check the sensor work fine, I just have to use the "python3" instead of "python" testing from command line:

pi@rpi2:~/Adafruit_Python_DHT/examples $ sudo python ./AdafruitDHT.py 11 26
Traceback (most recent call last):
  File "./AdafruitDHT.py", line 24, in <module>
    import Adafruit_DHT
ImportError: No module named Adafruit_DHT

But if I change the python command to python3:

pi@rpi2:~/Adafruit_Python_DHT/examples $ sudo python3 ./AdafruitDHT.py 11 26
Temp=26.0*  Humidity=37.0%

Anything I can do to force pythin3?  The DS18B20 works without any issues.

vitormhenrique, 2020-04-05:
plugin only works with python 2.7, python 3 is not supported, you need to install the libraries on python 2.7.

closed 2020-04-05 by vitormhenrique