vitormhenrique/OctoPrint-Enclosure #266: Possibility to run shell commands on GPIO input.

Snille, 2019-10-18:
**Is your feature request related to a problem? Please describe.**
Yes, not able to turn on the printer (and lights) with a physical button connected to a GPIO.

**Describe the solution you'd like**
When I push a button I would like to run a command in shell.
Example: I'm using this "usbrelay HURTM_1=1" command to turn on the PSU and the lights.
I would like to be able to do that via a button (GPIO input).

**Describe alternatives you've considered**
Using the G-Code "M80" and "M81" for on and off, but that's not in the menu either.

**Additional context**
Pretty please! :)

Best regards Erik

vitormhenrique, 2019-10-21:
the feature is already there, what you want is an OUTPUT with the shell script, and link that output with an GPIO input.

closed 2019-10-21 by vitormhenrique

Snille, 2019-10-21:
Ah, thank you! :)