vitormhenrique/OctoPrint-Enclosure #267: g-code command does not work in Cura "End g-code"

Vertigo1206, 2019-10-20:
I am using this plugin for controlling an external fan with a Rasperry Output.
Switching on/off via navbar works well, also when I send the gcode "ENC O1 S1" via Terminal, but I get the response "echo:unknown command"
[image]

But when I include the gcode in Cura, it does not work:
[image]
[image]
Is this because the printer is still busy with cooling?
In the Octopi log is nothing.

danioki, 2019-10-30:
leaving a comment here because I have the same problem but with the neopixel format of the G-code, the "ENC O2 R100 G100 B100" just send me the unknown command error.

vitormhenrique, 2020-04-05:
Do you have a gcode waiting for something before? for example cooling? have you tried removing that?

KriLL3, 2020-06-17:
I can't get the g-code control feature to work either, I can control an analog RGB LED strip using a driver connected to I2C with the color selector in enclose plugin tab, but putting "ENC O1 R255 G0 B0" in the terminal tab or in the starting g-code in prusa slicer, or entirely on it's own in an otherwise empty .gocde file does nothing beyond "Recv: echo:Unknown command: "ENC O1 R255 G0 B0"" I guess O1 is correct if the output is ID1?: https://i.imgur.com/0wv7dnB.png

SM-26, 2020-11-09:
Same here as well.
I'm trying to use G-code to change color of my LED strip, but I get an unknown command.

xconstarx, 2020-12-13:
Go to Enclosure Plugin Settings -> Scroll down an expand Advanced Settings -> Activate gCode Control -> Should fix your Problems ;)

[image]

ChristophDerndorfer, 2021-01-03:
As far as I understand it the "ENC" commands are custom commands by the Enclosure plugin and not part of the [official gcode set that the Marlin Firmware supports](https://marlinfw.org/meta/gcode/). Hence why the printer returns the "unknown command" message when OctoPrint sends it to the printer. Yet the Enclosure plugin listens for them and executes the corresponding actions.

And at least for me with the Neopixel Jewel everything works as expected.