vitormhenrique/OctoPrint-Enclosure #271: Raspberry 4

Kedryn, 2019-11-26:
**Describe the bug**
Can't make WS2812 work

**To Reproduce**
I'm running the latest octoprint and Enclosure plugin,
when i connect 3,3 and grnd to the a 16led circular, i get only first led green. Then i tried GPIO14 and GPIO18, both in Enclosure config and on the raspy4, but nothing changes.
Tried also a 4led linear, but all stays off on both pins.
Both works fine on a wemos d1 with tasmota.

**Expected behavior**
To light up

**Additional context**
[image]
[image]

cagott5, 2020-02-11:
Check in advanced menu that your on GPIO numbers or Pin numbers.  Just a guess

vitormhenrique, 2020-03-20:
@Kedryn did it work?