vitormhenrique/OctoPrint-Enclosure #274: Raspberry Pi 4 & DS18B20 sensors

dachew, 2019-12-04:
**Describe the bug**
I recently setup this plugin and configured 3 temperature sensors using new DS18B20's from Mouser (so I'm pretty sure they are legit).  On my Raspberry Pi 3, these sensors reported temperatures accurately (at least within 0.5/C of a thermometer).  However, I cooked the GPIO pins on my Pi 3 a few day ago and have replaced it with a Pi 4.  The temperature sensors are reading about 5/C to 7/C higher.

For example, right now the sensor on the external side of the enclosure is reading 22.1/C.  The room is about 14/C (it's in the basement and the heat is off).  The internal temperature sensor is reading 31.1/C, and cooling.  I had it warmed up for an ABS print.  I have a cigar box temp & humidity sensor mounted in the enclosure, and it's reading 26.6/C.

Previously, on the Pi 3, the enclosure sensor and the cigar box sensor were within 0.1/C of each other.  Which seemed impressive given these sensors are so cheap.

**To Reproduce**
Steps to reproduce the behavior:
1. On a Pi 3, configure a DS18B20.
2. Click on the temperature toolbar (if configured, or selected the enclosure tab).  Observe the temp.
3. Power off the Pi 3, remove the SD card, place it in your new Pi 4, and power up the Pi 4.
4. Observe the temperature reading again - it should be reading between 5/C and 7/C warmer.

**Expected behavior**
Temperature sensors should work the same regardless of Pi hardware versions.

**Additional context**
[image]

dachew, 2019-12-12:
Turns out a reboot or two seems to have resolved this.  So.  Weird.

Anyway.  Love the plugin!

arslankhawar, 2019-12-24:
sudo modprobe w1-gpio
pi@raspberrypi:~ $ sudo modprobe w1-therm
pi@raspberrypi:~ $ cd /sys/bus/w1/devices
pi@raspberrypi:/sys/bus/w1/devices $ ls
00-080000000000  00-700000000000  00-f00000000000  w1_bus_master1

/sys/bus/w1/devices/00-080000000000 $ cat w1_slave
cat: w1_slave: No such file or directory

and also in my code file following error is occuring.

   device_folder = glob.glob(base_dir + '28*')[0]
IndexError: list index out of range

Kindly someone help me to resolve this issue.It will be a great help for me in my final year project.

Thanks in advance ..!!!

dachew, 2019-12-24:
Did you follow the readme and install/enable the i2c, gpio, etc., for the Pi?

arslankhawar, 2019-12-25:
yes gpio and i2c is enabled.
Also changes the ds18b20 .
and now these files are showing
 ls
00-380000000000  00-780000000000  00-b80000000000  w1_bus_master1.

And error is same also

arslankhawar, 2019-12-25:
here is the code 

import os
import glob
import time
 
os.system('modprobe w1-gpio')
os.system('modprobe w1-therm')
 
base_dir = '/sys/bus/w1/devices/'
device_folder = glob.glob(base_dir + '28*')[0]
device_file = device_folder + '/w1_slave'
 
def read_temp_raw():
    f = open(device_file, 'r')
    lines = f.readlines()
    f.close()
    return lines
 
def read_temp():
    lines = read_temp_raw()
    while lines[0].strip()[-3:] != 'YES':
        time.sleep(0.2)
        lines = read_temp_raw()
    equals_pos = lines[1].find('t=')
    if equals_pos != -1:
        temp_string = lines[1][equals_pos+2:]
        temp_c = float(temp_string) / 1000.0
        temp_f = temp_c * 9.0 / 5.0 + 32.0
        return temp_c, temp_f
	
while True:
	print(read_temp())	
	time.sleep(1)

arslankhawar, 2019-12-25:
Here is the sensor i'm using and using pin 7 of pi zero w

[image]
[image]

dachew, 2019-12-26:
Looks good so far... Sorry for a maybe obtuse question, but I don't see power and ground connected.

arslankhawar, 2019-12-27:
i was using pi pins to give supply to sensor.

but now i'm using another 5v power supply to give voltage to sensor (well i just though may be from pi sensor is not getting enough voltage to active ) but error is same in both case.

Regards!
Arslan Khawar

dachew, 2019-12-27:
Chances are you've destroyed your sensors.  They are 3.3v only, and not 5v tolerant.  Also, you may have damaged your Pi's GPIO pin if the pullup resistor was bringing the input pin to 5v as well, but that could depend on a number of factors.  Be sure to only ever power a DS18B20 via a 3.3v supply, and never give your Pi more than 3.3v on any input pin.  Use a logic-level translator if you need to interface with a 5v-only device.

arslankhawar, 2019-12-30:
thanks for your help.
it works .

dachew, 2019-12-30:
That's great!

closed 2019-12-30 by dachew