vitormhenrique/OctoPrint-Enclosure #276: Can't run any script with the plugin

kliffom, 2019-12-07:
Hello, I'm trying to run a script using the plugin without any success. I made a few research without success, except one which had the same problem. I tried using his solution, but it didn't work.
Here are my scripts and what I did:
I wrote a simple Python script that turns high or low a pin on the GPIO. I know that I can do it easily with the plugin, but I want to be able to run that script also using the Telegram plugin, and I don't know if it's possible to "press the button" of the Enclosure Plugin with Telegram, so I decided to use the script. Here is the python script "LEDwhite.py":

`from os import path
import RPi.GPIO as GPIO
import time

LAST_STATE_PATH = "whiteLed.txt"
LED_NUM = 23

GPIO.setmode(GPIO.BCM)
GPIO.setup(LED_NUM, GPIO.OUT)

if path.exists(LAST_STATE_PATH) is False:
	# Crea un file per lo stato del led
	print ("File not exists. Creating...")
	f = open(LAST_STATE_PATH, "w")
	f.write("0")
	f.close

f = open(LAST_STATE_PATH, "r")
state = str(f.read())
# print state
f.close

if (state == '0'):
	print ("Turning LED on...")
	GPIO.output(LED_NUM, GPIO.HIGH)

elif (state == '1'):
	print ("Turning LED off...")
	GPIO.output(LED_NUM, GPIO.LOW)

else:
	print ("Unknown state. Turning off...")
	GPIO.output(LED_NUM, GPIO.LOW)

f = open(LAST_STATE_PATH, "w")
if (state == '0'):
	f.write("1")
elif (state == '1'):
	f.write("0")
else:
	print ("Wrong number. Setting default")
	f.write("0")
f.close
`

I added the permissions to everything with `sudo chmod 777 LEDwhite.py`, and if I run it using the shell it works without any problem. I then added it in the plugin settings, adding a new output and writing both "/home/pi/scripts/LEDwhite.py" and "~/scripts/LEDwhite.py", and when I click the button to run it, in the upper right corner I see a popup with a check, the word "Enclosure" and anything else, and the GPIO won't turn high.

I then thought that the plugin only runs shell scripts, so I wrote a simple bash script: 
`#!/bin/bash
python LEDwhite.py
` 
made it executable and added the permission like for the python script. I tried running it using the terminal and it works. I then added it to the plugin, but nothing again, just the popup with the check and the "Enclosure" word. 

In an example on the internet I saw that in the popup the plugin shows the output text of the script, so just to try I wrote a python script with a "Hello world" print and tried to run it. And as I expected, nothing again.

Now I just don't know what to do to run it. What am I missing?

GibranValle, 2020-08-08:
I have the same problem I also tryied with bash scripts. You should see in the log files if there is some problem, however when I look into my log file I cannot see any error. However the GPIO part writes this line:

2020-08-08 21:26:41,875 - octoprint.plugins.enclosure - DEBUG - Writing on GPIO: 23 value True

Xerony, 2021-04-22:
I have the same problem. I tryied python script and also bash script in terminal and everything works well but when I tryied it with plugin, nothing happened.