vitormhenrique/OctoPrint-Enclosure #277: Dangerous Heater Control Behaviour

wkellyoconnor, 2019-12-07:
**Describe the bug**
My setup includes relays for the lights, Heater control - (Enclosure heater), printer power and filament sensors. Printer is off, lights are off, enclosure heater is off.  I click on "Restart Octoprint" or "Reboot System" and the added enclosure heater turns on to 100F (This is what I typically run) even when autostart is off.  It just started this in the past few months. I don't reboot often. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to System 
2. Restart Octoprint or Reboot system.
3. During restart of Octoprint or a Reboot,  the Heater Control relay turns on and shows 100C as the setting even when it was off showing "off" in the drop down
[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/3935400/octoprint.log)

**Expected behavior**
Restarting Octoprint should not turn heater control on.

vitormhenrique, 2019-12-09:
please attach print screens of all your settings.

wkellyoconnor, 2019-12-17:
Sorry for the late reply messages like these go to Gmail's "forums" tab and
I don't check it much. While we are working this out I'll be sure to check
often. If you would rather all this go into GIT, let me know.

Kelly

*Screenshots. This is right after a simulated power outage. I had to
connect and shut off the heater control which was on 100F*
[image: image.png]
*Enclosure Plugin Page:*
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]

jcustenborder, 2020-01-01:
@wkellyoconnor Which relay shield are you using? Is it something like this [one](https://www.amazon.com/gp/product/B01FZ7XLJ4/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)? I've noticed for reference a few of these are wired incorrectly. Channel 1 on mine the NO/NC is fine, The other two channels are backwards.

wkellyoconnor, 2020-01-02:
I'd have to take the printer apart to see it but if my memory serves me, its a Sainsmart 4 channel relay board for arduino just like this one: https://www.sainsmart.com/products/4-channel-5v-relay-module . Oddly, it only does this when the mains power blinks and only started sometime this summer.

wkellyoconnor, 2020-01-08:
Has there been further thoughts as to what could be wrong? Every time I reboot the Pi, OR the power blinks here,  the heater relay kicks on. Is there another log, anything I could provide to help determine the cause? Again, when the pi boots the heater relay kicks on, I open Chrome, open the webpage, click the "Enclosure Plugin" tab the heater shows 100 degrees as a setting yet when it turn it off there is no indication the setting for that temp is stored anywhere. Where is that coming from? I don't even have to have the PC on in order to duplicate the issue. Please let me know how I can help you.

vitormhenrique, 2020-01-08:
I’m 100% sure that the issue that you are describing is caused by your wiring. 
Most likely the way that you are using the relay and it’s active open and active open / closed pins. When you reboot pi it will shut it’s pins off, so you are probably using the active high pin, when you should be using the active low.

wkellyoconnor, 2020-01-12:
I'll take the printer apart in the next week or so and take a look and switch stuff around to make it all active high. What you say makes sense but what I don't understand is I've used the printer with the enclosure heater for about a year before this issue came up. That puzzles me.

kami8607, 2020-01-18:
I had a similar behaviour when I did a fresh setup of octoprint a while ago. After startup it always would set the heater to 40°C. After I removed this setting from config.yaml while octoprint was shutdown, the problem was gone.

vitormhenrique, 2020-04-05:
@wkellyoconnor do you still se this issue? have you tried deleting all configuration and creating it again?

wkellyoconnor, 2020-04-06:
@vitormhenrique, I checked all the wiring as you suggested. Everything is in order. Oddly it stopped doing it. Recently, I updated to the latest Octoprint and also added a small UPS to the Raspberry so power blinks won't affect it anymore anyway. I'd say for now you can just close this, forgive me for not responding sooner.

vitormhenrique, 2020-04-06:
don't worry, sometimes is difficult to track bugs, and some times adding or deleting inputs on a certain way can create weird configurations. 
I'm gonna add a way to export and import settings on the next release, maybe that will help troubleshooting.

closed 2020-04-06 by vitormhenrique

bumbalu, 2020-06-25:
Hi. I can confirm the same behavior: despite heater options "start with server" and "Auto start-up" are off Heater is turned on right after Octoprint start or reboot. It is set to 46 C at which I some day have printed something. As mentioned above I will try to go to config.yaml and find and remove this setting. The wiring in my case is 100% correct. But it is really dangerous for somebody who is not aware of such behavior. In my case, for example, printer is located in other room.