vitormhenrique/OctoPrint-Enclosure #279: Unable to install; setup tools missing

victoriatraindoc, 2019-12-16:
Tried to install a new setup of "Enclosure Plugin" on a brand new install of Octopi . Got the following error:
"No module named setuptools"
I then ran "sudo apt-get install python-setuptools"
and then "sudo python setup.py install" and it installed beautifully.
Hope this helps

Jyde314, 2020-04-05:
Thank you ! I was stuck at this step... Now it works :
Finished processing dependencies for Adafruit-DHT==1.4.0
pi@octopi:~/Adafruit_Python_DHT $ cd examples
pi@octopi:~/Adafruit_Python_DHT/examples $ sudo ./AdafruitDHT.py 22 4
Temp=22.4*  Humidity=32.9%
BUT in Octoprint the "Enclosure" plugin still give me (even after refresh/reload) :
| Temperature | Humidity
0°C | 0%

Jyde314, 2020-04-05:
Solved by this one : https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/291

vitormhenrique, 2020-04-05:
Thanks for coming back and documenting!
Question, what version of python / octoprint are you using?

closed 2020-04-05 by vitormhenrique

Jyde314, 2020-04-22:
Hi, sorry for delay to answer this. Version of python is 2.7.13 Octoprint is up to date so 1.4