vitormhenrique/OctoPrint-Enclosure #284: AM2303 showing 0 °C and 0 %

EnYo15, 2020-01-22:
The temperatures in shows 0 °C and 0 %

sudo ./AdafruitDHT.py 2302 16 is giving me proper response and i disabled sudo to run python commands in the advanced options. Jet the log sais 

octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.

how can i solve this problem

vitormhenrique, 2020-01-23:
read the log....

octoprint.plugins.enclosure - INFO - Failed to execute python scripts, **try disabling use SUDO on advanced section of the plugin**.

EnYo15, 2020-01-23:
I disabeld sudo in advanced settings and stilll get this Message!

vitormhenrique, 2020-01-23:
try restarting your pi....

EnYo15, 2020-01-23:
I restarted with both Software Options, I repluged the power and in reinstalled the Plugin.
The only thing I did not, was to reinstall OctoPrint

EnYo15, 2020-01-25:
now my octoprint.log file gives two errors

2020-01-25 11:07:10,339 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2020-01-25 11:07:10,341 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('need more than 1 value to unpack',)

again I unchecked Use Sudo in the advanced options and i restartet my pi multiple times

vitormhenrique, 2020-04-05:
Have you figured this out?

Try disabling sudo password:

```
sudo visudo
and add pi ALL=(ALL) NOPASSWD: ALL to the end of the file.
```

EnYo15, 2020-04-20:
I did try this as well.
I had to reinstall Octoprint and now it works.

closed 2020-04-20 by EnYo15