vitormhenrique/OctoPrint-Enclosure #287: Invert PWM duty cycle

gergaly, 2020-02-05:
**Is your feature request related to a problem? Please describe.**
Due to my control logic (an extra transistor in the PWM signal) I would like an inverted PWM signal. When I link the PWM output to temp sensor it will be activated when the print starts (if I understand correctly). This means my fan will rotate at maximum frequency without even printing.

**Describe the solution you'd like**
The actual PWM duty cycle on the GPIO should be 100 minus the wanted duty cycle.

**Describe alternatives you've considered**
I thought I'll do the fan control outside of OctoPrint in a bash or python script run by cron or run as a daemon.

4lunester, 2020-03-26:
Excellent idea when using PWM  fans, makes for easier hardware integration.

insaneFu, 2020-04-29:
If you use an optocoupler instead of a transistor, you can hookup its input between 3.3V and your PWM GPIO. (I have PC817, don't forget a resistor, in my case 470O.)

That way, there is no need for inverting the PWM signal.