vitormhenrique/OctoPrint-Enclosure #293: Enclosure Plugin is blank after Fresh Install

txbone66, 2020-02-29:
**Describe the bug**
A clear and concise description of what the bug is.
I'm new to Raspberry and Octoprint, so I'm sure it's an Ido3t error, but to offset your frustration, I've sent some $ to buy a pint.  Octopi is running, but when I install the Enclosure Plugin from the plugin manager, this is the screen that I see.  
[image]

I've uninstalled, reinstalled, and rebooted, but I get the same result.  I'm wanting to use this to send a GPIO command to a power block that will turn the printer and lights on and off.

[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/4270473/octoprint.log)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add screenshots of your settings screen, so I know how to recreate the data. Also enable the debug information under the advanced section of the plugin, reproduce the issue and attach the [octoprint log](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299) here.

txbone66, 2020-02-29:
like I said, Ido3t error.

closed 2020-02-29 by txbone66

vitormhenrique, 2020-02-29:
I guess you found the configuration screen? ;)

txbone66, 2020-02-29:
Yep!  something simple!  I swear I spent a long time reading and checking, but all this is so new to me, I just missed that.