vitormhenrique/OctoPrint-Enclosure #294: Can I command SG90 servo?

GpMissi, 2020-02-29:
**Is your feature request related to a problem? Please describe.**
I would like to use servo to move the camera (Tilt/Pam) and focus on different areas, 

**Describe the solution you'd like**
is it possible to do this with the current program? how you do it?

tks

(Scusate per il pessimo inglese, uso il traduttore di Google)

GpMissi, 2020-03-01:
Similarly, can I create a button to turn on / off or reset the printer by closing a relay for a few seconds?

Tobi0892, 2020-04-05:
I am also very keen on controlling an SG90 servo via PWM. 

[Here](https://www.reddit.com/r/3Dprinting/comments/a3r619/octoprint_controlling_a_servo_via_gpio/), someone reported that he was unsuccessful due to unprecise movement. Is there someone else who tried it?

vitormhenrique, 2020-04-05:
I might have an idea on how to implement, but for now, a work around would be have two PWM outputs on the two servos, and using the city cycle to change the servo position.

1.5 milliseconds (ms) pulse width should center the servo, 2.0, should move it all the way to one side and 1.0 to the other.