vitormhenrique/OctoPrint-Enclosure #299: PWM Based upon Temperature not working or not understood by me

jessaskey, 2020-03-18:
Firstly, thank you for writing this plugin, it really does so much!

I have a PWM Output configured which drives a Fan via a MOSFET on the PWM line.

When I have it NOT linked to the Temperature Sensor (DHT-22) then it works as expected... a default value of 100 will turn the fan OFF and a setting of 0 will turn it full ON. I can control this in the UI by setting a value and hitting the checkmark and this works great.

Now, when I link it to my Temperature sensor (which correctly shows a temp of 23 degrees C and updates properly in the Enclosure UI), things start to get wonky for me. 

Here is what I am trying to accomplish and then I will review my settings.

1. Temp < 25 degrees C = FAN DEFAULT (100 - Off)
2. Temp @ 25 degrees C = Start doing linear interpolation  = FAN @ 100
3. Temp @ 30 degrees C = End range for linear interpolation = FAN @ 0
4. Temp > 30 degrees C = FAN @ 0 (full speed)

I have my settings as follows

[image]

So, right now since my enclosure is at 23 degrees C, then I would expect my fan to stay off because it is lower than the Temp A number, however as soon as I save this configuration, my FAN turns ON. 

Maybe I have mis-understood the configuration for this? Could you please advise when you have a chance?

I can include more info for sure, I just didnt want to go down that path unless you wanted me to.

thank you!

jessaskey, 2020-03-18:
A little update on this.... I noticed this in a few other posts regarding PWM too and I think I have some info that might help clarify this 'inversion' issue on the PWM outputs.

As you note above, I had my fan being OFF with a PWM value of 100 (which seems inverted), well that is because it was, I was trying to run my 4-pin fan like a two pin fan through a MOSFET like this...

https://www.sensorsiot.org/variable-speed-cooling-fan-for-raspberry-pi-using-pwm-video138/

However, Im dumb and I realized that I should be just doing this instead which is the correct way to drive a 4 pin PWM fan with a RPi...

https://www.raspberrypi.org/forums/viewtopic.php?t=244194&start=25

Now, with the direct wiring, my settings are no longer inverted. So I can have settings like this...

[image]

However, with that tweak, I still have the same issue... but at least the FAN is not defaulting to FULL ON now (due to me removing the inversion issue)...

So, my behavior is now that the FAN is OFF by default (PWM = 0) and as I raise the temperature past 23 degrees, my fan is not increasing in speed tho. As above, I can see the Enclosure temperature increasing, but the fan does not seem to react to it. If I remove the temperature control linking to the PWM, I can manually set the PWM to 0 (OFF) and PWM to 100 (ON) just fine however.

thoughts?  thank you!

vitormhenrique, 2020-03-20:
I'll do some investigating ....
I started a masters on AI and haven't had much time to work on this plugin...  I started a major port and changes on it but I estimate that would be probably a month worth or changes....

Meanwhile you might just to control the fan as a simple IO with triggers starting and finishing the print, or maybe controlled using g-codes with your slicer...

jessaskey, 2020-03-20:
Yeah, Im still learning on my setup what may be best. Even with a PWM of 0, my fan still seems to spin at a low speed, so I am probably going to have to move to a tandem Relay on Power Pin + PWM signal for speed. In the end, you are right, there may be other ways to approach this because right now Im still learning the pros and cons of each approach just in Enclosure. In the end I just wanted to have my enclosure attempt to stay at 72°F (22.2°C) and have the Exhaust fan ramp up to 100% somewhere around 80°F (will have to figure that out in the end)... 

But for right now, even with Debug Logging turned on, Im not really getting any info dumped from Enclosure about what the temperature is and how it is setting the PWM value. Maybe I will be a good human and fork your code and add in a bunch of debug logging at least so I can figure out what is going on... there is a 51% chance Im doing something wrong, so that isn't your responsibility anyway. :-)

Flyingmatdes, 2020-04-24:
Hello Vitormhenrique, first thanks for this plugin, it is really well done. However, I have the same trouble with PWM when linked to Temp Sensor. My Temp sensors (I use two) are just working fine my fan is working great when I do select the PWM duty manualy. But when its linked to the temp, the fan never starts !

I read that you where really busy and I do understand. May I suggest that when you fix this issue, you also add the possibility to start the temperature interpolation not only when a print starts, but also at any moment ! I would like to have my cooling fan running as soon as I start Octoprint !

insaneFu, 2020-04-29:
If you use an optocoupler instead of a transistor, you can hookup its input between 3.3V and your PWM GPIO. (I have PC817, don't forget a resistor, in my case 470O.)

That way, there is no need for inverting the PWM signal.

insaneFu, 2020-04-29:
However, I noticed that for some specific PWM frequencies, the voltage / brightness / PWM output at the used GPIO is not proportional to the duty cycle percentage, as would be expected.

10Hz - everything works fine
100Hz - non-linear behaviour
1000Hz - everything works fine again, but the fan obviously doesn't like it
specifically unsettling: 25kHz (the specified PWM freq for PC fans) - non-linear behaviour

I am in the process of testing right now. 23kHz seems to work voltagevise, but I still need to test the fan itself.

I suspect, this hat either something to do with the plugin's PWM generation code or the Rasberry Pi (3B+ in my case) hardware.

Any ideas?

insaneFu, 2020-04-29:
> I read that you where really busy and I do understand. May I suggest that when you fix this issue, you also add the possibility to start the temperature interpolation not only when a print starts, but also at any moment ! I would like to have my cooling fan running as soon as I start Octoprint !

Amen to that!