vitormhenrique/OctoPrint-Enclosure #300: Neopixel circuit dont work

taka054, 2020-03-28:
Im new in the raspberry scene and has a prusa i3 and control it now with octoprint. 
I also want to lighten up my prints at the night and buyed W2812B LED circuits and my first connection is running after 3 hours of trying. 

I connected everything directly on the rpi 3B+ and when i test it directly with a pyton test script it works perfectly and i can set the color that i want. this is my test: 

**import board
import neopixel
pixels = neopixel.NeoPixel(board.D21, 12)
pixels.fill((255, 0, 0))**

after save, i start it with:  python3 test.py and it works

thanks for helping

BothFamily, 2020-04-03:
Sorry - I can't help you, but I'm having a similar issue. I can control the NeoPixel lights directly with Python - but nothing seems to work in Enclosure. I've tried every pin (entering both board # and BCM #), fed the stick external power etc.

I'm not sure what else to try, I've cleared the GPIO, started with a fresh copy of the plugin, even went so far as to reload a fresh copy of OctoPi.

I'm using a pi 4B with 2GB of ram.

taka054, 2020-04-04:
ok, good to know that im not alone with this problem. 
Right at the moment, i start the neopixel before print. Im looking for a workaround to control the neopixel with a script, controled over the octolapse plugin.

haithamsh77, 2020-04-04:
I'm having the same issue. I can use the following code to get the LED Strip to work, but not through the plug-in:
```
import board
import neopixel
pixels = neopixel.NeoPixel(board.D18, 52)
pixels.fill((255, 0, 0))
```
I can also use the code documented on the plug-in gitHub page: octoprint_enclosure/**neopixel_direct.py** and it works after changing the values in the code to match my setup of course.

The above tests confirm that LED strip works on the RPi and that the necessary libraries are working. Currently testing it with a fresh install of Octoprint and another RPi (and will not update Octoprint to V 1.4, to see if the update caused this!). Will post an update...

My setup is as follows:
-  RPi 3 B+
- LED Strip Data pin on GPIO (BCM 18).
- LED Strip (100 LEDs), and powered by a separate 15A/5VDC adaptor.
- RPi and LED Strip shared GND.
- RPi Octoprint Version 1.4.0 all updates installed.

@taka054 please share the workaround to control the neopixel. Thanks.

haithamsh77, 2020-04-04:
I couldn't get the script to run from within the plugin UI. Tried everything I could think of, nothing worked! Running a python script from the command line worked, but not from the plugin UI.

I then decided to update Octoprint to V 1.4. I then followed the following instructions from the plugin instructions page. With that, the "neopixel-direct" option worked! Not the script though. I think the main problem was the "Sudo" thing, so it is very important to do as it says below:

> For Neopixel
If your setup does not have pip install pip: sudo apt-get install python-pip

> Install the required library: sudo pip install rpi_ws281x

> rpi_ws281x really needs sudo, and you need to setup up so your rpi does not ask for a password > when runing a python script, so run:

> sudo visudo

> and add pi ALL=(ALL) NOPASSWD: ALL to the end of the file.

> Also backlist the audio kernel:

> sudo nano /etc/modprobe.d/snd-blacklist.conf

> add the blacklist snd_bcm2835 to the end of the file

@taka054 @BothFamily Let me know if the above works for you.

Good luck!

BothFamily, 2020-04-04:
I've unfortunately done all that, still cannot get the plugin to trigger the lights... I even re-did everything, in case I missed a step. 

Still functions manually using a script similar to yours though.

haithamsh77, 2020-04-04:
@BothFamily when you run the script from the command line, which one the following works for you:

sudo python3 xyz.py
OR
sudo python xyz.py 

(i.e. "python3" or "python" or Both)? In my case, previously only "python3" worked. I noticed that the plugin worked when I could use both "python3" and "python"

taka054, 2020-04-04:
i already installed everything before, nothing changed. 
When i try to start the test with sudo python, it doesnt respond, only with python3, i dont know why.

BothFamily, 2020-04-05:
Sorry - haithamsh77, I’m in the middle of a 30 hour print, tomorrow I will try it. To be honest, I’ve only tried using python3 - as that was what was written in the adafruit guide. Based on Taka’s comments, I’m guessing I will have the same issue.

Thanks for all your suggestions.

vitormhenrique, 2020-04-05:
you have multiple python environments on your system, you need to install and get it working with **python** not python3.

This plugin only supports python 2.7 for now.

taka054, 2020-04-05:
> you have multiple python environments on your system, you need to install and get it working with **python** not python3.
> 
> This plugin only supports python 2.7 for now.

ok, than we found the error, thank you. 
I followed a guide that used python3 and this enclosure plugin, but i cant reach him to ask some questions how they do it work.

BothFamily, 2020-04-05:
So I confirmed I have Python 2.7.16 installed, along with Python 3.7.3 

When I run the test.py script - I am getting an error using just python that I don't get with python3. It's unable to import module - line 1.

I did notice this in the adafruit guide:

"If your default Python is version 3 you may need to run 'pip' instead. Just make sure you aren't trying to use CircuitPython on Python 2.x, it isn't supported!"

I'm just not certain why, following the guide - I get to this point, nor where to look - to finish trouble shooting.

Paul

vitormhenrique, 2020-04-05:
this plugin does not work with CircuitPython.

you are trying to install and use a newer library that this plugin does not use.

BothFamily, 2020-04-14:
So I tried from scratch on a new card, with no luck. Heck, I even gave up and tried some other plugins - all to no avail. I stumbled upon this [https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/](url).

I finally have my NeoPixel's glowing from within octoprint - rather than just using code as above. Do each of the steps - even the one where it says "we aren't concerned with those..."

No I just need to get the lights to actually do as they are told, rather than changing colour on manual input.

Namain, 2021-02-02:
I was having the same issue described above, but I managed to get my Enclosure Plugin's NeoPixel Direct setup to work by removing the link from /usr/bin/python to /usr/bin/python2 and creating a new one to /usr/bin/python3. This has some potential to break other things, namely anything on your system that depends on Python2 specifically and assumes that the python command is pointing to Python2, but it appears to work for me.

1. SSH into your Octoprint instance
2. Run the command 'sudo rm /usr/bin/python'
3. Run the command 'sudo ln /usr/bin/python3 /usr/bin/python'

PS1TD, 2022-07-10:
> you have multiple python environments on your system, you need to install and get it working with **python** not python3.
> 
> This plugin only supports python 2.7 for now.

Thank you! I had python3 and didn't even realize I had to have python 2 installed.
After installing python2 and the rpi_ws281x library with pip I was able to control the LEDs.

jifop, 2023-11-30:
> I was having the same issue described above, but I managed to get my Enclosure Plugin's NeoPixel Direct setup to work by removing the link from /usr/bin/python to /usr/bin/python2 and creating a new [...]

This also worked for me