vitormhenrique/OctoPrint-Enclosure #302: [Feature REquest] Python 3 Compatibilty

jneilliii, 2020-03-31:
**Is your feature request related to a problem? Please describe.**
Running the latest version of OctoPrint 1.4.0 in a Python 3 environment does not allow for the use of this plugin as it is detected as incompatible.

**Describe the solution you'd like**
The ability to run the plugin in a Python 3 environment.

**Additional context**
Please see [here](https://docs.octoprint.org/en/master/plugins/python3_migration.html) for additional information regarding OctoPrint and Python 3 Migration.

vitormhenrique, 2020-04-05:
Hi Idus,
What is the plan for python 3? when is gonna be released? is it gonna support python 2 and 3?

If so I have no idea how I'm be able to support both python 2 and 3.

I'm using some libraries that only support python 2, and other libraries that support python 3 for the temperature sensors don't support python 2.

Any idea how can I handle this?

jneilliii, 2020-04-05:
The current 1.4.0 version is both python 2 and 3 compatible. It's been mentioned a couple of times on the forum and YouTube by Gina (foosel) that python 2 support will probably be dropped in a year or so. The best thing to do is look for alternative modules to replace those that are not both 2 and 3 compatible.  The link above in the original post has some great information in regards to what you'll need to do and links to helpful resources for the most common conversion issues.

vitormhenrique, 2020-04-05:
Yeah,

It's been a while since I worked on the plugin and followed octoprint development. 

For me is gonna be realllyyyy difficult to support both python 3 and 2. Most of the sensor libraries that I use are python 2 only. And all new libraries for those sensors are python 3 only. Unless I write my own libraries to read sensors I will probably not be able to support both.

I wanted do drop python 2 and change all sensors to use circuit python Adafruit library, it would make my life so much easier and allow expanding my plugin to be used on many more sensors.

jneilliii, 2020-04-05:
Eventually it will be Python 3 only, maybe creating a separate plugin that is Python 3 only would be the easiest route forward. I guess another option would be to create a "middle man" module that wraps the different options based on compatibility, but that seems like it would be a pain. I personally would go the first route and fork your current version into a new repository and use the newer python 3 modules and set the compatibility to python 3 only.

tsayles, 2020-06-20:
@vitormhenrique Do you have a list of the "sensor libraries that I use are python 2 only"?  

_BTW -- I like the functionality the Enclosure plug-in enables and want to help keep it viable long term. Maybe I can help track down a couple Python 3 compliant alternatives._

jneilliii, 2020-06-20:
Another option dependent on the available libraries you've found is using
```
try:
    import py2library as gpio
except ImportError:
    import py3library as gpio
```
Only trick there is the methods/properties or whatever you call them past the gpio part would need to match or you have to add additional try/except blocks throughout your code, or conditionally flag a boolean for what version of Python you're using and add a bunch of if/else blocks within your code. 

In my opinion, I think you should just port it to Python 3 only plugin and if anyone complains, direct them to this post to upgrade their OctoPrint to Python 3.

https://community.octoprint.org/t/upgrade-your-octoprint-install-to-python-3/21232

vitormhenrique, 2020-06-20:
plan to migrate for a python only plugin, I think all sensors are available using adafruit micro python libraries that run on python 3, but I suffered an arm injury last Sunday and broke both bones on my left forearm, it will be impossible to work on on this plugin in the near future, I'm struggling to make it work with a full time job + masters in computers science.

jneilliii, 2020-06-20:
Ouch, I wish you a swift recovery @vitormhenrique. My carpel tunnel has been acting up since I've been sheltered in place so I know a very small bit of what you're dealing with.

vitormhenrique, 2020-10-14:
Ok, after a while I'm finally ready to start working on this plugin again...

First I designed a PCB with some nice features that will allow me add new features and expand this plugin on the near future.... 

This will allow me test new stuff a little bit more easily and will probably use those shields on my enclosure...

It's basically a raspberry pi shield with the following features:

- PCA9635 PWM driver with 6 GPIO's broken out (control servos, led's and fans without need to use raspberry pi really poor pwm)
- ADS1015 12 bit ADC with two channels broken out, to be able to use thermistors as temperature inputs...
- Raspberry Pi fan connected to the PCA9635 driver (to be controlled based on rPI CPU temperature)
- 1 5V level shifted output to control neopixels
- Bunch of extra GPIO outputs / input to be used, most of them with a convenient ground to be used as an input button, and some with 5V and 3.3V that can be used with external relays

I ordered the parts, now need to solder some PCB's and test the circuit...

[image]

psonoda, 2020-10-16:
Looks like a fun project, thanks for working on the P3 upgrade. Enclosure is one of my few "must have" plugins for Octoprint.

haplm, 2020-10-20:
Very nice board, I was thinking about doing something similar, glad that I don't need to refresh my Eagle skills :-)

Can you please add a relay, pretty please? That would be extremely helpful, as I could make an update of my octopi box, and include your board:
https://www.prusaprinters.org/prints/41225-octoprint-box-raspberry-pi-4b-waveshare-43-dsi

vitormhenrique, 2020-10-20:
There is absolutely no space on the board to add relays on it... 

**BUT** I added two connectors specially designed to drive two 2 Channel DC 5V Relay Modules, so total of 4 relays... the pinout has everything on them with a JST connector.

Another reason why I went this route is that I like to keep my high-voltage stuff far from where my fingers can touch... and by having relays on the board there would be high-voltage near them...  

I ordered 5 boards and soldered one to start developing the new release, everything it's working, it's a 4 layer board, could not get all traces on a 2 layers... 

[image]

haplm, 2020-10-20:
I see, it is a hat. I can't say I like hats, but since there's everything, it is probably OK. What is the standoff height to mount it?

And fait point with the relays, connector work without any issue. Is it possible to get a soldered board later on, or will I have to solder all the SMD stuff myself? ;-)

vitormhenrique, 2020-10-20:
I just like that it's simple and easy to mount, minimize cables passing around and wanted to be fool proof.... 

My solder past was spoiled and I had to solder everything with a soldering iron for my initial test...  I have to get some more solder paste and do a proper soldering... With all those connectors and 4 layer board this is not going to be a cheap board for me to manufacture and stock parts, I'll create a separate thread to see if people are interested, if so, how many, and try to figure out pricing.... 

I can see that this might be useful for robotics and other projects as well but again, as a hobby assembling this and testing would be very time consuming...

vitormhenrique, 2020-10-20:
official hardware discussion: https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/358

vitormhenrique, 2021-02-05:
Just a quick update: progress has been slower than I expected for the next release, I have been focusing on COMPLETELY re-writing the plugin. When I first started this plugin it was my first project on python, now I have a lot more experience as I actually switched jobs and became a python software developer.

But with the new job, breaking my arm last year, and starting a master in computer science life got busy realllll quick.

Most of the changes so far were UI related, I want to make the plugin easier to configure as most of the issues opened on GitHub are configurations that get corrupted and are a little difficult to setup really. 

Another goals that I have on the new version:
- Including all necessary libraries to all sensors, so users don't need to install libraries manually, My plan is to have Adafruit circuit python installed on the requirements, that probably has all libraries already.
- I want to have some object oriented stuff (enclosure outputs, and inputs) making easier to extend the plugin in the future.
- Switched how I store the configuration from a big dictionary to an array of " enclosure" objects, this is gonna make easier to debug and troubleshoot configurations.
- I want to be able to export and import configurations to a json file from the browser, to share, help, etc....

If anyone is interested on helping we can divide work, and get this working faster.

Sneak peak on the new UI:

[image]
[image]
[image]

n3roGit, 2021-04-27:
Great work. I am looking forward to the new version. Can I already test it somehow and give feedback?

chimcen, 2021-05-29:
Hi,

awesome project - but i‘ve a question about the topic… are there any news to tell? Is there a release date for enclosure plugin with python3 support?
I‘d love to see your great plugin running again in my environment :)

Adjuro, 2021-06-07:
Hi,

Thank you very much for your hard work !
Your plugin is a must-have for me and I am really hopping you will be able to release an update soon.

From my side I use an other PCB type I found on Hackaday. It seems to be a solution with less connectivity that yours but it is ok for my needs (Relay, Fans, On/off switch, Temperature and humidity ,Leds).

I am working on building a case with all integrate functions (Cooling for the RPi, interface to the "outside". Try to have it as small/compact as possible but it is really challenging (over heating issues).

Due to latest Octopi updates, my sensor is not working properly but it was expected.

My dream would be to have an deeper integration of enclosure plugin to Octodash. But that is not the place to discuss it.

[image]

[image]

[image]

[image]

[image]
[image]
[image]
[image]

jneilliii, 2022-08-24:
Plugin works with Python 3.

closed 2022-08-24 by jneilliii