vitormhenrique/OctoPrint-Enclosure #304: How do I set the "initial" temperature for a temperature control output?

spiff72, 2020-03-31:
Can someone provide some input on how to set the initial value for a temperature control that is already set up? 

I have a control set for an exhaust fan that maintains temperature, and it appears to default to a temp of 27C, but I can't figure out how to change this value.  I am able to change it via the custom Gcode that I have set up for my printer (material specific), and I can adjust the temperature for "end" gcode so that sets the value as I want it (i would like to make it 30C), but when i restart octoprint, it is reset to 27C.

I tried adding the GCode that I use for the end of print to Octoprint's "when initially connecting to printer", but this caused the Octoprint interface to get messed up badly (formatting went crazy - the top status bar was messed up, etc).

vitormhenrique, 2020-04-05:
The only way to set initial temperature for temperature control is to use custom enclosure g-code commands described here:

https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/G-CODE-Control

Enable g-gode controls on the advanced settings and follow the wiki.

closed 2020-04-05 by vitormhenrique

spiff72, 2020-04-05:
I used the custom enclosure Gcode it caused odd issues in my Octoprint interface.

I used this code:
ENC O2 W0 S29 ; set enclosure temp to 29 degrees

Does it make sense to place it in the GCODE scripts part of octoprint, in the "After connection to printer is established" section?

I will try it again after the print I have running now is completed.

vitormhenrique, 2020-04-05:
I would put this on your slicer to run before all other scripts.

spiff72, 2020-04-05:
Yes - I already do that with PrusaSlicer - it sets the enclosure temp based on the temperature that the tool is set to (I set enclosure temp higher for non-PLA materials).  The end-of-print script for my printer sets the enclosure to 29C now. 

I was just looking for a method to set the temperature so it is correct after restarting Octoprint (or the printer) prior to sending any prints.  I had thought it was built into the enclosure plugin, but I was mistaken (hence my post here).

spiff72, 2020-04-05:
As I mentioned above, the formatting of the web interface for Octoprint flaked out when I tried this the first time, but maybe there was something else happening that caused that.  

I will try again after the print I am running is done (about an hour) and will follow up with the results.  I will screen shot the interface if it is messed up again and post it here.

Thanks!

spiff72, 2020-04-06:
Well I tried the above and this time it worked fine!  Strange - I don't know what was causing the problem last time, but I am all set now.
Thanks!