vitormhenrique/OctoPrint-Enclosure #305: Python 3 compatibility

Idus, 2020-04-01:
I was just asked by OctoPrint today to do a Python 3 test (there is a new app for that) and report an issue to all plugin githubs that were found incompatible with Python 3 (if not already such an issue was stated).
Did the test and got back that the your plugin was Incompatible with Python 3 and could not find any other open issue about it - so here it is. :)

Cheers

vitormhenrique, 2020-04-05:
Hi Idus,
What is the plan for python 3? when is gonna be released? is it gonna support python 2 and 3?

If so I have no idea how I'm be able to support both python 2 and 3.

I'm using some libraries that only support python 2, and other libraries that support python 3 for the temperature sensors don't support python 2.

Any idea how can I handle this?

vitormhenrique, 2020-04-05:
I just noticed that there was already one issue:

https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/302

closed 2020-04-05 by vitormhenrique