vitormhenrique/OctoPrint-Enclosure #306: DHT11

renorr3d, 2020-04-02:
I'm totally new to this so I im missing something certainly obvious.*
I can't get my DHT11 reading. 
I think I am not installing the adafruit library the right way, if someone can help..
I connect to my pi via ssh with putty
then I run the command specified in the readme, and the library seems to install

cd ~
git clone https://github.com/adafruit/Adafruit_Python_DHT.git
cd Adafruit_Python_DHT
sudo apt-get update
sudo apt-get install build-essential python-dev python-openssl
sudo python setup.py 

but when I try to test the library with this command

cd examples
sudo ./AdafruitDHT.py 11 4

I have this answer 
-bash: cd: examples: no such file or directory

And I cant get any reading in octoprint (it display 0 / 0) even if I uncheck sudo in advanced settings.
Can someone please help ?
Thanks

vitormhenrique, 2020-04-05:
I'm sorry, 
But this is not an octoprint-enclosure issue, you need to get the library installed and working first.

Unfortunately I'm too busy with my masters to help you getting the library installed, it should be pretty simple. But it looks like it's not even installing the library, hence **examples: no such file or directory**

Get the library installed and working, after that we can help you.

closed 2020-04-05 by vitormhenrique

renorr3d, 2020-04-12:
Hi Vitor,

Thanks for your reply, I know I was missing something..
For the noobs like me with the same problem here is the steps I missed

sudo su
sudo apt-get install python-setuptools
sudo apt-get install python3-setuptools

git clone https://github.com/adafruit/Adafruit_Python_DHT
cd Adafruit_Python_DHT
sudo apt-get update
sudo apt-get install build-essential python-dev python-openssl
sudo python setup.py install

Cheers