vitormhenrique/OctoPrint-Enclosure #311: WS2812B can't control brightness

Piranit, 2020-04-18:
Hi.
I'm using Raspberry Pi 4 Model B 4GB
I connected Din pin to 18 GPIO.
LED works, I can change color.
**But I can't control brightness.**
Even when in settings I set brightness parameter to zero, LED still shining at the maximum brightness.
Can anybody suggest something?

Piranit, 2020-04-19:
Today I have tested with `strandtest.py` from `rpi_ws281x` library and the changing of brightness works perfect.
But why doesn't it work from plugin?

viatsyshyn, 2020-04-19:
Hi, have similar problem

blackote, 2020-04-21:
Hi Guys, so I just had to re-setup my Pi using the RPi 4.  Just make sure to follow the Neopixel instructions on the readme.  then reference to Issue number 172 [https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/172#issuecomment-410864599](url)

adding the DMA 10

Piranit, 2020-04-25:
@blackote thanks for your time, but the problem is related to **BRIGHTNESS** control. The solution of the problem you've recommended won't help. Why?

I've already figured out the problem, the problem is that the parameter `LED_BRIGHTNESS` is not used anywhere at all.
You can check it out [here](https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/octoprint_enclosure/neopixel_direct.py) and **LED_DMA** is already working well, so you don't need to create the `LED_DMA` variable by yourself.

How to solve the problem with the brightness?
You have to replace 
`strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT)` 
in `octoprint_enclosure / neopixel_direct.py` with 
`strip = Adafruit_NeoPixel (LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS)`
Then brightness control will work. 

I've tested this way and everything works fine. I think now it's easy to understand, that the `LED_BRIGHTNESS` parameter haven't been used and that has created the problem.

I've made a [pull request](https://github.com/vitormhenrique/OctoPrint-Enclosure/pull/314) about this problem, hope that it'd be taken into account soon.