vitormhenrique/OctoPrint-Enclosure #315: Outputs triggering twice?

1Ghasthunter1, 2020-04-27:
Just installed enclosure plugin, and am running switch that is 5V when filament is in, GND when out. Enclosure picks up the signal and executes my output, which does the following:

`M117 Message1
G28
M117 Message2
G4 S1
`

However, the output sequence gets triggered twice. What could be happening?

vitormhenrique, 2020-05-26:
maybe switch is having some bouncing issues? maybe triggering on rising and falling edges? 
not enough information to help out, sorry...

mrwizerd, 2020-07-13:
You could possibly add a debounce circuit on the switch to deal with the issue.  This is a common problem with switches, you could probably look up debounce on google to see if something there might solve the issue.