vitormhenrique/OctoPrint-Enclosure #319: RC servo control,, and  enclosure temp view on temperatures page

jaygrovr, 2020-05-05:
Wanted to know if you can do some RC servo control through your interface.  I would mainly be wanting to use this to control some vent openings, so that I can cool down my enclosure after a print, so the servos would want to get turned on a move to a set position.  This could be gcode.  I dont have any outputs on my controller board for servos at this point.

I also wanted to know if you could put the temperature of the enclosure on the temperatures page, so that it would be easy to monitor and graph temps of the enclosure when using heating or cooling.

vitormhenrique, 2020-05-26:
https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/294

vitormhenrique, 2020-05-26:
https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/34

vitormhenrique, 2020-05-26:
please search issues before opening new ones, that is common best practice on GitHub, we spend a lot of time writing free code for you, and only request doing some 2 minute search before asking the same question again.

closed 2020-05-26 by vitormhenrique

jaygrovr, 2020-05-26:
Yes I did, but wasnt exhaustive enough.  Sorry for the extra work.
Jason

vitormhenrique, 2020-05-26:
no worries, well, follow those threads with updates whenever I have the change they will be implemented.