vitormhenrique/OctoPrint-Enclosure #320: Trigger PSU Plugin via API

Tobias0044, 2020-05-11:
I would like to execute a shell command as output to activate the PSU plugin via API. However, the entire website freezes until I kill the process via SSH.
My command: curl -s -H "Content-Type: application / json" -H "X-Api-Key: 1AFD0CXXXXXXXXXXXXXXXXXXXX1B11" -X POST -d '{"command": "togglePSU"}' http: // octopi / api / plugin / psucontrol

If I execute the command via SSH everything works.
I tried to do it in a .sh file, but the same thing happens

vitormhenrique, 2020-05-26:
logs? print screens? 

my best guess as you didn't provide the requested documentation when opening the issue, and probably didn't look at the closed issues as well: your command takes a while to execute and it locks the execution of the plugin, you can't do that, you need to execute the command on the background, most times you can just add an "&" at the end of the command, sometimes you need to pipe result somewhere.

closed 2020-05-26 by vitormhenrique

Kerjox, 2020-09-12:
Hi, I have had the same problem when I try to use PSUcontrol and Enclosure with the script:

> `curl -s -H "Content-Type: application / json" -H "X-Api-Key: 1AFD0CXXXXXXXXXXXXXXXXXXXX1B11" -X POST -d '{"command": "togglePSU"}' http://0.0.0.0/api/plugin/psucontrol`

If you are **testing the scritp whith the buttons of the octopi interface**, it makes hang the web interface.
[image]

But don't worry, your script is not wrong, simply the button of the web interface is completely broke.

So the **way to make it work** is testing it with a pysical button, and surprisingly, it works perfectly.

Screenshot to the config of the script:
[image]

And the Input associated:
[image]
[image]
You can also see the electrical schema in this thingi [here](https://www.thingiverse.com/thing:4024711)

And of corse, the script:

`curl -s -H "Content-Type: application/json" -H "X-Api-Key: 1AFD0CXXXXXXXXXXXXXXXXXXXX1B11" -X POST -d '{ "command":"togglePSU" }' http://0.0.0.0/api/plugin/psucontrol 
echo 'Botón de encendido pulsado manualmente'`

This script make a notification in your octopi web interface that says "Botón de encendido pulsado manualmente":
[image]