vitormhenrique/OctoPrint-Enclosure #323: Possible to add i2c GPIO Expander option? MCP23017 chip.

JaredC01, 2020-05-15:
Hello,

I'm in somewhat of a unique position.  I'm using a Pimoroni Hyperpixel 4.0 display, which uses all of the GPIO pins for the display, making all of the regular GPIO pins unusable.  I've got an i2c GPIO expander board, based on the MCP23017 chip, which connects via i2c.  I can manually control the GPIO via the command line, however I can't find any way to make use of the MCP23017 chip inside of the Enclosure Plugin, and when attempting to add ANY GPIO pin, it breaks the display.

Would it be possible to add support for this chip?  The command-line GPIO control has built-in extensions for the 23017 chip, however the Python library doesn't seem to, as best I can tell.

If this is possible, please let me know!  I would be happy to test anything you can come up with.

vitormhenrique, 2020-05-26:
Possible? yes!
but if you follow some of my comments here I started a master degree and haven't had time to work on this plugin in a while.
I have a lot of stuff planed and started making some changes but it would need completely re-writing most of the plugin and I don't have the time now, so don't expect this soon...

JaredC01, 2020-05-26:
I understand.  Thank you for the reply, and good luck with your masters degree!