vitormhenrique/OctoPrint-Enclosure #325: Open raspberry pi terminal

yaron30109, 2020-05-21:
Hi,

I am new to raspberry pi so I hope this question is not too funny . I installed Octoprint and the enclosure plug in, but when it says " Open raspberry pi terminal and type:" I am not sure what to do next, 
How do you open the raspberry pi terminal? 

I am using raspberry pi zero w, and DHT11 temperature sensor. 

Thanks.

I-like-stuff, 2020-05-21:
You need to use ssh to access your raspi’s terminal here is a guide:
https://www.raspberrypi.org/documentation/remote-access/ssh/
Once you have connected login in with user pi and the password, default is raspberry, and then type in what the instructions say.

You could do it connected to a monitor but you might as well learn to use ssh as it can be very useful especially for convenience as it is much less messing around with ssh.

yaron30109, 2020-05-21:
I only have Octoprint on the Raspberrypi, should I install anything else?

I-like-stuff, 2020-05-21:
No nothing else needs to be installed, you just need the IP address of the pi or if your device that you are accessing from supports bonjour then you can use octopi.local

yaron30109, 2020-05-21:
Access the Octoprint is not a problem. I already installed the OctoPrint-Enclosure plugin.

 My question is how do I access the raspi’s terminal  from Octoprint to to install Adafruit library to use the temperature sensor?

theopensourcerer, 2020-05-21:
@yaron30109 That's what @I-like-stuff said - you need to "ssh" into your Raspberry Pi. Read the link he posted.

yaron30109, 2020-05-21:
The issue is that I'm not sure how to communicate with the Raspberry Pi other than the Octoprint platform, or open the Raspberry Pi Configuration page. I guess I will have to do a bit more digging on this.

I-like-stuff, 2020-05-21:
Do you know the IP address of the raspberry pi, this is what you can use to ssh into the pi?

theopensourcerer, 2020-05-21:
@yaron30109 When someone says a rasberry pi "terminal", they mean the command line (like when you run cmd.exe in Windows)...

ssh is a protocol that lets you connect from your desktop PC or laptop (where your web browser is) to the Raspberry Pi so you can enter commands into the "terminal". ssh gives you a remote "terminal" to your Raspberry Pi.

Read this link and the subsequent ones. It will explain how to set this up.

https://www.raspberrypi.org/documentation/remote-access/ssh/

yaron30109, 2020-05-21:
@I-like-stuff - know the IP address of the raspberry pi, I use it to enter the octoprint page.

Thanks @theopensourcerer, I will have a look.

yaron30109, 2020-05-21:
So If I can access Octoprint by typing the IP address, SSH is not already enabled?

I-like-stuff, 2020-05-21:
SSH should be enabled by default.

yaron30109, 2020-05-21:
@I-like-stuff Much appreciated all the help. Maybe I didn't described my question properly at the beginning. 
I followed the instruction on the OctoPrint-Enclosure page, and  I was not sure how to do the next step. Where do I copy paste the command lines below? In the Ocotprint server page? 
"
You need to install Adafruit library to use the temperature sensor on raspberry pi.
Open raspberry pi terminal and type:

cd ~
git clone https://github.com/adafruit/Adafruit_Python_DHT.git
cd Adafruit_Python_DHT
sudo apt-get update
sudo apt-get install build-essential python-dev python-openssl
sudo python setup.py install

Note: All libraries need to be installed on raspberry pi system python not octoprint virtual environment.
"

I-like-stuff, 2020-05-21:
You copy those commands when you have connected via ssh to the pi, on Windows you can use putty instructions here:
https://mediatemple.net/community/products/dv/204404604/using-ssh-in-putty-(windows)

 You enter the IP address, user name(pi) and password(raspberry as default) 

Download putty here:
https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html

yaron30109, 2020-05-21:
@I-like-stuff  Thank you for the detailed reply.  I was able to see sensor readings on the PuTTY terminal, but no on Octoprint. I will try to reinstall the plugin.

yaron30109, 2020-05-21:
Before installing the Adafruit library, the readings on Octoprint were 0, now I get °C and % with no numbers.

vitormhenrique, 2020-05-21:
Disable sudo on the advanced section of the plugin configuration.

yaron30109, 2020-05-21:
@vitormhenrique , @I-like-stuff , much appreciated for all the help, finally can see all readings after deleting and installing again the plugin.  Great plug in, thanks.

closed 2020-05-21 by yaron30109