vitormhenrique/OctoPrint-Enclosure #326: Can't contril pin 28

albebert, 2020-05-25:
**Describe the bug**
need to use pin 28 to control simple relay but can't error in octoprint log : 

 octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('The channel sent is invalid on a Raspberry Pi',)

**To Reproduce**
Steps to reproduce the behavior:
add an output on gpio pin 28 

*Additional context*

tested with gpio command  : 

gpio mode 31 out
gpio write 31 1
gpio write 31 0

and all is working

oblivionratula, 2020-05-26:
Well 27 is the highest GPIO pin, isn’t it? So as the error states, 28 is invalid. If you mean physical pin 28, that’s GPIO 1 according to what I’m looking at, and that may conflict with ID_SC.

But I’m no expert.

albebert, 2020-05-26:
yes i speak about pin N° 28;

but when i use the gpio from cli all is working so i d'ont understand why with enclosure plugin they don't work

vitormhenrique, 2020-05-26:
please, follow instructions on the troubleshooting guide... and instructions on how to open an issue.

Asking for questions with no log, no print screens does not help and makes impossible for anyone to help...

[image]

vitormhenrique, 2020-05-26:
https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/Troubleshooting-Guide#open-a-new-issue

vitormhenrique, 2020-06-20:
you are globally using incorrect names for the pin based on your message, read instructions or follow the guide on how to ask queens, closing for now

closed 2020-06-20 by vitormhenrique