vitormhenrique/OctoPrint-Enclosure #327: Octoprint Crashes after Print

Kotentopf1337, 2020-06-07:
Based on Log, I'd assume its something with the plugin.
```
2020-06-07 11:45:19,886 - octoprint.util.comm - INFO - Finished in 14572.932 s.
2020-06-07 11:45:19,889 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2020-06-07 11:45:19,893 - octoprint.filemanager.analysis - INFO - Starting analysis of local:MyPrint.gcode
2020-06-07 11:45:19,901 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/MyPrint.gcode
2020-06-07 11:45:19,966 - octoprint.plugins.PrintJobHistory - INFO - Print result:success, CaptureMode:successful
2020-06-07 11:45:19,972 - octoprint.plugins.PrintJobHistory - INFO - Start capturing print job
2020-06-07 11:45:19,958 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: MyPrint.gcode, owner: <User>
2020-06-07 11:45:19,992 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Start parsing Slicer-Settings
2020-06-07 11:45:20,008 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2020-06-07 11:45:20,118 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Finished parsing Slicer-Settings
2020-06-07 11:45:20,698 - octoprint.plugin - ERROR - Error while calling plugin enclosure
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_enclosure/__init__.py", line 1573, in on_event
    file_name = os.path.basename(payload["file"])
KeyError: 'file'
2020-06-07 11:45:20,698 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintDone
```

vitormhenrique, 2020-10-19:
are you still having this issue?

rucker, 2021-02-16:
I am seeing this issue as well (same message/Traceback as above). Two interesting facts in my case:
1. So far I have only installed the OctoPrint-Enclosure plugin (haven't changed any settings yet) on two identical Ender 3 Pros
2. Of my two printers, I'm only seeing this exception on one of them