vitormhenrique/OctoPrint-Enclosure #330: UI Improvement suggestion: nested configuration of inputs and outputs

KriLL3, 2020-06-16:
It's very cumbersome to configure the plugin when you have multiple inputs and outputs. It turns into a long confusing to navigate single vertical list.

It would be a lot more user friendly to instead use a more "nested" approach, make the main list just each item's "label" with all it's configuration nested inside clicking either a wrench icon next to the label, the label itself, or expand the contents between labels like a collapsing menu item, there are many different ways to do it but they all boil down to a nested approach that eases navigation.

Custom control editor is a good example of how to do similar complex configuration easy to manage.

vitormhenrique, 2020-06-20:
this would be veryyyyy low on my huge list of improvements, I personally like the current way and think its intuitive.
I also suffered an arm injury last Sunday and broke both bones on my left forearm, it will be impossible to work on on this plugin in the near future, I'm struggling to make it work with a full time job + masters in computers science.

KriLL3, 2020-06-20:
Yikes, hope you mend soon, I don't find the current setup intuitive at all though, especially having to hunt for the "add outputs..." button hidden somewhere randomly in the very long list of inputs and outputs, if you have 1-2 of each it's manageable but if you load up a lot it turns into a very long piece to scroll though.