vitormhenrique/OctoPrint-Enclosure #331: Instructions for adding AM2302 support

ritchiedc, 2020-06-25:
**Describe the bug**
When I followed the steps to add AM2302 support the last step, "sudo python setup.py install", failed because it could not find the setuptools.

**To Reproduce**
Steps to reproduce the behavior:
1. Start with a new Octopi image on a raspberry pi 3B
2. Install the OctoPrint-Enclosure and add an AM2302 sensor
3. Follow the instructions in the Readme.md for an AM2302 sensor

> cd ~
> git clone https://github.com/adafruit/Adafruit_Python_DHT.git
> cd Adafruit_Python_DHT
> sudo apt-get update
> sudo apt-get install build-essential python-dev python-openssl
> sudo python setup.py install

4. The final step fails to find setuptools

**Expected behavior**
The "sudo python setup.py install" step should successfully execute.

**Additional context**
The problem can be corrected by adding installation of python-setuptools to the line

> sudo apt-get install build-essential python-dev python-openssl

as follows

> sudo apt-get install build-essential python-dev python-openssl python-setuptools

P.S. Sorry to hear about your broken arm

vitormhenrique, 2020-06-25:
Thanks,

this was documented here: 
https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/317

but I didn't get a chance to put it in, and one programing is not quite working....

will address in a future release.

meganssmith, 2020-08-14:
I am unable to get this to work :(  It even fails the sudo ./AdafruitDHT.py 2302 4 returning Failed to get reading. Try again!,  I did try installing the new Library from AdaFruit as well, and that works in the command line, so I know the sensor works and is wired up right.  I am using a Pi4 with 4gigs on it.  Any Ideas where to go next?

Thanks

ritchiedc, 2020-08-14:
I will try to help. Try testing without the “sudo”, i.e. “./AdafruitDHT.py 2302 4”

Let me know if that works

Ritchie Christian

meganssmith, 2020-08-14:
Thanks Ritchie for helping.  I tried it and got 

Traceback (most recent call last):
  File "./AdafruitDHT.py", line 41, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/common.py", line 94, in read_retry
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/common.py", line 81, in read
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/Beaglebone_Black.py", line 213, in read
RuntimeError: Error accessing GPIO. Make sure program is run as root with sudo!

vitormhenrique, 2020-08-14:
there are a bunch of issues with those old Adafruit libraries and python versions newer than 3.0 ...

as I mentioned on other issues, I'm planing to migrate this plugin to python 3.0 only, but I broke my arm couple month ago and  only got better now, with masters and full time job behind I have not had any change to work on this plugin.....

I suggest if you want to use this plugin install a really old version of octopi around Version 1.3 and stick with it until I got newer versions working....

ritchiedc, 2020-08-14:
That’s interesting. I can’t run mine as root. It will not reliably read the AM2302 and will sometimes give me the error you saw. I have a Raspberry pi 3B. I don’t know if that is a difference. I do have one question. Did you install Octoprint on Raspberry Pi OS or did you start from Octopi OS with Octoprint already installed?

Ritchie Christian

meganssmith, 2020-08-14:
I started with Octopi OS with Octoprint already installed.  I also used the Neopixels in Enclosure Plugin and that works great.

meganssmith, 2020-08-14:
Yes, sorry to hear about the arm, I am sure that would make coding a pain.  I also understand the time commitment issues.  I can live without the Temp for now, I primarily got the plug in for the neopixel support.  

> there are a bunch of issues with those old Adafruit libraries and python versions newer than 3.0 ...
>
> as I mentioned on other issues, I'm planing to migrate this plugin to python 3.0 only, but I [...]

ritchiedc, 2020-08-14:
Since I can’t help much more on the Linux side let’s at least double check the electrical side. There are different pin numbering schemes. I would like to verify that starting from the corner nearest the connector you have connected the AM2302 as follows:

2nd pin (5V) in the row next to the edge is connected to +5v
3rd pin (Gnd) in the same row is connected to Gnd
4th pin (GPIO 4) in the second row is connected to the Signal

Ritchie Christian

meganssmith, 2020-08-14:
YEs, I think that is right.  If I use the new AdaFruit Lib (python3) it works and reports.

> Since I can’t help much more on the Linux side let’s at least double check the electrical side. There are different pin numbering schemes. I would like to verify that starting from the corner [...]

ritchiedc, 2020-08-14:
It could be a difference between the 3B and 4B. I am using the latest Octpi (0.17.0) and Octoprint (1.4.2).

Sorry I can’t provide any more help since I’m too cheap to buy an new 4B.

Ritchie Christian

meganssmith, 2020-08-14:
It could be, and in the big picture, it is not that big of a thing to me.  I appreciate you stepping up an giving it a shot all the same. 

Thanks

> It could be a difference between the 3B and 4B. I am using the latest Octpi (0.17.0) and Octoprint (1.4.2). Sorry I can’t provide any more help since I’m too cheap to buy an new 4B. Ritchie Christian

ritchiedc, 2020-08-14:
I have a hard time letting this kind of stuff go. One last question. Did you install the adafruit libraries using the octoprint-enclosure instructions or using the adafruit instructions? I want to make sure we used the same process.

Ritchie Christian

meganssmith, 2020-08-14:
I started with the instructions in the enclosure.  I then tried it with the modification listed in this thread, and I tried it with adafruit directions last.   

> I have a hard time letting this kind of stuff go. One last question. Did you install the adafruit libraries using the octoprint-enclosure instructions or using the adafruit instructions? I want to make sure we used the same process. Ritchie Christian

ritchiedc, 2020-08-14:
Let’s try one more thing. Go to the Enclosure settings where you set up the sensor and neopixels (wrench -> scroll down and select enclosure plugin). Scroll down to the bottom of the enclosure settings and select “Advanced options”. Then uncheck “Use SUDO” and see if you start getting temp and humidity readings. I had to do this to make it work. If it works check to make sure the neopixels still work. I don’t hold out a lot of hope since your system seems to need “sudo” to work but it can’t hurt to try.

Ritchie Christian

meganssmith, 2020-08-15:
So now I get 32F (or 0C) and 0 Humidity.  Neopixels did not work with SUDO off.  Turned Sudo back on, Lights work again, incorrect temp stayed.  

> Let’s try one more thing. Go to the Enclosure settings where you set up the sensor and neopixels (wrench -> scroll down and select enclosure plugin). Scroll down to the bottom of the enclosure [...]

ritchiedc, 2020-08-15:
Sorry I couldn't help. It's still puzzling why it works without root for me but you need root.

galoki, 2020-10-20:
Just adding my tests to the thread for feedback and assistance in troubleshooting.  Not expecting any support on this unless you happen to know a fix, I can wait til Vitor is healed and ready.  

I have both PI 3b and 4b.  with the 3b I can get it to work in the CLI using python3.  Enclousure plugin doesn't work.  I move the sd card and connections for the sensor (wired as diagramed) to the Pi 4b and it wont read the sensor in the cli and the plugin doesn't work.