vitormhenrique/OctoPrint-Enclosure #334: Using REST api to get input gpio state

5thLabRat, 2020-07-21:
Does anyone know if it is possible to use the REST api to check the current state for a gpio input?

I'm currently using a mechanical switch to detect the end of the filament but I would like to be able to monitor this switch from my Home Assistant server. 

If I use the list specific input endpoint I get the correct json response but none of the entries appear to be the actual state.

mrath, 2020-12-08:
Did you resolve this? I’m in the same boat - would also like to exposure temperature readings through homebridge, but no luck so far.

vitormhenrique, 2020-12-08:
i just tested here using 
```
http://**{URL}**/plugin/enclosure/getOutputStatus?apikey=**{APIKEY}**
```

and got reply with:

```
[{"status": false, "index_id": 6}]
```

what endpoints are you trying to use?

mrath, 2020-12-08:
I am not sure what you mean. We are trying to get the state of an input. 
I managed to use the api to get a long, seemingly general-purpose JSON with the input data (input is index 1):
`http://**{URL}**/plugin/enclosure/inputs/1`
... so I guess I just have to find a way to get temp and humidity from that JSON into homebridge.

martinezpenya, 2020-12-23:
I'm facing same problem, I want to get temp sensor from enclosure plugin through REST to put it in Home Assistant.
If I use the old API documentation as @mrath suggested I get 404 error....
the getOutputStatus with apikey from @vitormhenrique is woking well.... but... how I get temp and humidity from my dht22 attached sensor??

steviehs, 2021-02-23:
Hmm... I am rather confused now: 
I am running in exactly the same issue while trying to use the API for Octodash:

   `http://**{URL}**/plugin/enclosure/getOutputStatus?apikey=**{APIKEY}**` 

works, but

` http://**{URL}**/plugin/enclosure/inputs`

gives me a 404.  Running 4.12 ....

What am I doing wrong? How are the both API specs (in Wiki and the md file in the repo) related?