vitormhenrique/OctoPrint-Enclosure #335: intermitant gpio buttons

tonywright, 2020-08-02:
Hi, I have the Enclosure plug in all set up on a Raspberry Pi 4, 4Gb. Pressing a button (Connected between +3.3 and gpio pi via a resistor, on pin 17 BCM) only works on "rare" occasions. I can see the gpio input is changing every time I press the button  by using "GPIO readall" on the command line, but octoprint does not respond very often.

Obviously the connections are correct because it does occasionally work if I sit there constantly pressing the button.
On the rare times it does work, it then sometimes works multiple times for one click.

I am running the latest version of the plugin, 4.13.1, the raspberry pi is only running octoprint. The command the button is sending is a g-code for "Home" (I have 4 other buttons set up and they all behave with the same indifference to me pressing them)

I have turned on the debugging mode but nothing seems to appear in the logs except the command (when it does pickup the gpio change)

Any ideas if there is anything else required to get this to work correctly?

Thanks,

Tony

vitormhenrique, 2020-10-19:
I must say that this is probably because of wiring as I never seen this issue.... 
are you using pull up / down resistors?
have pictures of your wiring?

thestephenmarshall, 2020-11-20:
This is not likely a wiring issue. Looks like there were some errors related to threading:

```
2020-11-20 15:45:41,543 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-11-20 15:45:41,551 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2020-11-20 15:45:41,552 - octoprint.server - INFO - OctoPrint 1.4.2
2020-11-20 15:45:41,558 - octoprint.plugin.core - INFO - 26 plugin(s) registered with the system:
| !Access Anywhere - The Spaghetti Detective (1.3.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_thespaghettidetective
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Arc-Welder (1.0.0+u.bb71e8f) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_arc_welder
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
| !Detailed Progress (0.2.5) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_detailedprogress
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Enclosure Plugin (4.13.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_enclosure
|  Error Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking
|  File Check (2020.8.7) (bundled) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_file_check
|  Firmware Check (2020.9.23) (bundled) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmware_check
|  Firmware Updater (1.7.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmwareupdater
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Login UI (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/loginui
|  Macro (0.3.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_macro
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Themeify (1.2.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_themeify
|  TouchUI (0.3.16) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2020-11-20 15:45:41,581 - octoprint.environment - INFO - Detected environment is Python 2.7.16 under Linux (linux2). Details:
|  hardware:
|      cores: 4
|      freq: 1200.0
|      ram: 917016576
|  os:
|      id: linux
|      platform: linux2
|  plugins:
|      pi_support:
|          model: Raspberry Pi 3 Model B Rev 1.2
|          octopi_version: 0.17.0
|  python:
|      pip: 19.3.1
|      version: 2.7.16
|      virtualenv: /home/pi/oprint
2020-11-20 15:45:41,581 - octoprint.server - INFO - ------------------------------------------------------------------------------
2020-11-20 15:47:05,390 - octoprint.server.util.flask - INFO - Passively logging in user stephen from ::ffff:192.168.1.212
2020-11-20 15:47:05,393 - octoprint.access.users - INFO - Cleaning up user session A065206215B64858ACE8A30580234B86 for user stephen
2020-11-20 15:47:05,396 - octoprint.access.users - INFO - Logged out user: stephen
2020-11-20 15:47:05,399 - octoprint.access.users - INFO - Logged in user: stephen
2020-11-20 15:47:06,261 - octoprint.server.util.flask - INFO - Passively logging in user stephen from ::ffff:192.168.1.212
2020-11-20 15:47:06,264 - octoprint.access.users - INFO - Cleaning up user session A065206215B64858ACE8A30580234B86 for user stephen
2020-11-20 15:47:06,267 - octoprint.access.users - INFO - Logged out user: stephen
2020-11-20 15:47:06,269 - octoprint.access.users - INFO - Logged in user: stephen
2020-11-20 15:47:06,959 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.1.212
2020-11-20 15:47:07,005 - octoprint.server.util.flask - INFO - Passively logging in user stephen from ::ffff:192.168.1.212
2020-11-20 15:47:07,006 - octoprint.access.users - INFO - Cleaning up user session A065206215B64858ACE8A30580234B86 for user stephen
2020-11-20 15:47:07,007 - octoprint.access.users - INFO - Logged out user: stephen
2020-11-20 15:47:07,009 - octoprint.access.users - INFO - Logged in user: stephen
2020-11-20 15:47:09,234 - octoprint.server.util.sockjs - INFO - User stephen logged in on the socket from client ::ffff:192.168.1.212
2020-11-20 15:47:13,658 - octoprint.printer.standard.job - INFO - Print job deselected - user: n/a
[... cut ...]
2020-11-20 19:37:22,888 - octoprint.server.preemptive_cache - INFO - ... done in 0.66s
2020-11-20 19:37:22,943 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.1.212
2020-11-20 19:37:22,983 - octoprint.server.util.flask - INFO - Passively logging in user stephen from ::ffff:192.168.1.212
2020-11-20 19:37:22,984 - octoprint.access.users - INFO - Logged in user: stephen
2020-11-20 19:37:25,095 - octoprint.server.util.sockjs - INFO - User stephen logged in on the socket from client ::ffff:192.168.1.212
```

thestephenmarshall, 2020-11-20:
Restarting Octoprint resolves the issue.

jacopotediosi, 2025-05-01:
Hi everyone, the recently merged PR #504 should have fixed this issue. Could you please reinstall the plugin and, if confirmed, close the issue?