vitormhenrique/OctoPrint-Enclosure #337: Filament Sensor

Puffymania, 2020-08-12:
Hey,
writing because i can't get the Filament sensor to work.
I'm using a simple 3 pin micro switch to detect filament.
Wired like this NC connectors wired to GPIO 24 and C to ground right below GPIO 24.
Just wanna make sure how i should set up Enclosure Plugin?
Best regards Puffy.

chimeranzl, 2020-09-23:
Make sure you're in BCM mode. BCM vs BOARD mode have different pin numbers.  Test by manually by ssh to Octoprint, then from cmd line run: 

gpio readall

Now trigger the microswitch (hold it closed) and run the command again. Compare the V column to see which one is different. The BCM corresponding this is the number you want to use.  If none of the binary outputs change, then check your wiring.

vitormhenrique, 2020-10-19:
@Puffymania have you tried the suggestions? 

still have issues? can you attach pictures of the setup, schematics on how are yo wiring with screen shots of the configuration being used?

Puffymania, 2020-10-26:
Hello and sorry for the late response, just became a dad and it was a difficult birth, but all is good now.
I actually got it figured out with a little trial and error testing. 
But thanks for the response.
Nice to see a community so willing to help us noobs.
Great day to you all.

closed 2020-10-26 by Puffymania