vitormhenrique/OctoPrint-Enclosure #338: "sudo apt-get update" failed

JBoe-62, 2020-08-24:
Hello

I apologize for placing this issue here because it's not directly a bug of octopi. But I hope, here is someone who can solve the problem.

I tried to install the software for a DHT22 sensor acc. to [https://github.com/vitormhenrique/OctoPrint-Enclosure]

As mentioned in the headline, everything worked fine but the last command "sudo python setup.py install" failed and ended in the following failure message:

Traceback (most recent call last):
  File "setup.py", line 1, in <module>
     from setuptools import setup, find_packages, Extension
ImportError: No module named setuptools

I tried it several times, so a typo is not the reason.

I'm new to Linux and RaspberryPi without any experience. Any help from your side is appreciated.

- Linux octopi 4.19.75-v71+ #1290
- OctoPi version: 0.17.0

vitormhenrique, 2020-08-24:
As you are new to GitHub as well, just letting you know that before opening any issue, is costumary searching for that issue before....
as everything here is free, out time the the only thing that cost some money, and the less that I can spend answering questions twice the more time available to put on the software:

https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/279

https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/317

if you can't get it to install I'm sorry but it is not an octoprint enclosure issue.....

closed 2020-08-24 by vitormhenrique

JBoe-62, 2020-08-24:
Sorry for wasting your time, but I had read all the reported issues titles before I wrote this.
The title of #317 is too general. Title #279 is much more clearer. I must have overlooked that. Sorry again for that.
For your information:
With the additional command line "sudo apt-get install python-setuptools" and switching off sudo command in the Enclosure Plugin it all works.
Thanks for your support and continue to get well soon.

vitormhenrique, 2020-08-24:
no worries man!
and it's not wasting my time! I want to help as many people as possible.... and I enjoy having my software being used and helping other people, just a quick letting you know, I don't think people, specially new on linux / GitHub / open source need to know everything, it's quite an journey to get started on this and a good one on the long run!