vitormhenrique/OctoPrint-Enclosure #339: Enclosure Plugin using deprecated payload descriptor

mmcp42, 2020-08-28:
the Plugin has not been updated to work with OctoPrint 1.4.1
it is using **"file"** to obtain the file name
it needs to use **"name"**

**To Reproduce**
Steps to reproduce the behavior:
print anything
examine octoprint.log
you will see (for example)
`2020-08-28 15:50:27,119 - octoprint.plugin - ERROR - Error while calling plugin enclosure Traceback (most recent call last): File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 230, in call_plugin result = getattr(plugin, method)(*args, **kwargs) File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_enclosure/__init__.py", line 1573, in on_event file_name = os.path.basename(payload[`**"file"**`]) KeyError: 'file'`

**Excerpt from 1.4 documentation:**

**PrintDone**
A print completed successfully.
**Payload**:
**name**: the file’s name
**path**: the file’s path within its storage location
**origin**: the origin storage location of the file, either local or sdcard
**size**: the file’s size in bytes (if available)
**owner**: the user who started the print job (if available)
**time**: the time needed for the print, in seconds (float)
**Deprecated since version 1.3.0:**
**file**: the file’s full path on disk (local) or within its storage (sdcard). To be removed in 1.4.0.
**filename**: the file’s name. To be removed in 1.4.0.

**Expected behavior**
clearly the error should not appear

the fix is to change **"file"** to **"name"**
I have tried this and it no longer throws the error

hope that all makes sense 

also hope your broken arm recovers soon
if I can help . . . do let me know
thanks
Mike

vitormhenrique, 2020-09-15:
would you minding forking, fixing and sending a pull request?

mmcp42, 2020-09-16:
All done

Hope your arm is on the mend

Cheers

 

 

Mike McPherson

07711 238858

mmcp42@yahoo.com

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus