vitormhenrique/OctoPrint-Enclosure #340: Can only turn on GPIO (can't switch it off afterwards)

nickbuhling, 2020-08-31:
Dear developer,

Thanks for the great plugin, it is very useful for managing my enclosure. 
I found a little issue when using an IO output to control a GPIO connected to a relay to switch on/off the enclosure's lighting. I am able to use the navbar button to switch the GPIO on, but afterwards I cannot switch it off again. This has worked for me in the past so I don't know why it is not working now. 
Do you have any idea or suggestions?

Kind regards,
Nick

psonoda, 2020-09-17:
Hey Nick, I have a light and a fan on micro ss relays and they work great with the enclosure plug in. Both on and off. Have you put a volt meter on the gpio pin to see if it's turning off? What kind/model of relay are you using?  Have you tried a different gpio pin?

nickbuhling, 2020-09-18:
Thanks for your reply! After I stopped using GPIO 1 and GPIO 0 for the relays the feature started to work perfectly again. Therefore I will close this issue.

closed 2020-09-18 by nickbuhling