vitormhenrique/OctoPrint-Enclosure #341: Confirmation popup on action

Pietrogramma, 2020-09-07:
Hi, compliment for excellent plugin, I use it and I found it perfectly.
It's possible have a popup for the confirmation on an action? 
I use the plugin with the GPIO2 and GPIO3 in a RASPI4, for power ON/OFF two relay; one where I connected the power supply of the printer and one where I connected the power for a  strip led.
It's possible insert a popup that say me to confirm the power ON or OFF of the LEd or Printer?
For example: I connect to my Octoprint server with a telephone (small screen). I want to turn on the light to check the status of the printer. I think I'm touching on my screen the Plugin voice of the casing that turns on the LED, but I'm wrong and I touch instead the command that turns off the printer and I combine a trouble. Is it possible that the plugin asks me to confirm if I want to turn off or ON the printer or the LED?
Thanks and ciao

PS sorry my bad english

TazerReloaded, 2020-12-02:
Had the same idea, I have a special ATX-PSU with 24V output and control it with a transistor on a GPIO pin. Switching on and off works like a charm, but if I accidentally hit the wrong button, the power can get cut mid print.
A simple prompt (Really toggle <output name> off? <Cancel> <Confirm>) would be enough, and the feature must be activated with a checkbox in the settings.

marklieberman, 2021-02-12:
I also want to request this. I have relays to control the printer power, external lights, and fans. I am always extra nervous when switching lighting on/off during a print. It is particularly dangerous when I'm using TouchUI on my phone and a mis-touch could be disastrous.