vitormhenrique/OctoPrint-Enclosure #346: BME280 not working

JanHBade, 2020-09-17:
Hello,

[image]

[image]

LogFile:
2020-09-17 14:55:02,488 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2020-09-17 14:55:02,488 - octoprint.plugins.enclosure - WARNING - An exception of type TypeError occurred on log_error. Arguments:
("a bytes-like object is required, not 'str'",)

Sensor is working (tested via SSH)

roomme13, 2020-09-17:
Same issue, turn on / off "use SUDO" option didn't bother

mik-at, 2020-09-24:
Same Issue with a DHT11 here.

f-hnt, 2020-09-29:
I'm experiencing the same issue with SI7021. The sensor is working properly, but the Enclosure plugin is showing zeros for the temperature and humidity. 
The problem appeared after migrating from Python2.7 to Python3.8. I've tried rolling back to Python2 and it worked properly again.
"Use SUDO" option doesn't affect the behavior. 

It seems to me related to the "bytes type" that was introduced in Python 3, but I don't know how to solve it.

vitormhenrique, 2020-10-12:
have you tried the sudo thing? also as @f-hnt mentioned, plugin only supports python 2.7 at the moment.... working on python 3 now....

vitormhenrique, 2020-10-19:
@JanHBade update on this?

f-hnt, 2020-10-20:
@vitormhenrique Do you have a rough idea when Python3 support will be ready?

vitormhenrique, 2020-10-20:
Started this week, will probably take couple weeks to maybe month....

I'm completely re-writing the plugin, not re-using a single line of code, and there are probably 10k lines of code on the old plugin....

So don't expect anything too soon...

f-hnt, 2020-10-22:
> 
> 
> Started this week, will probably take couple weeks to maybe month....
> 
> I'm completely re-writing the plugin, not re-using a single line of code, and there are probably 10k lines of code on the old plugin....
> 
> So don't expect anything too soon...

No pressure. 
Thanks for the effort. Looking forward to it. :)

oeijoenj, 2020-11-16:
so DHT11 will not work on phyton 3 as well ?

Will-It-Work-42, 2021-01-03:
Hello evrerybody,

I'm facing the same issue : my BME280 is working fine, it is correctly seen by the system : 

```
root@octoprint:/home/pi#  i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- 76 --
```

Values are correctly reported by the python script : 

```
/home/pi/bme280.py
Chip ID     : 96
Version     : 0
Temperature :  14.74 C
Pressure :  1010.70826168 hPa
Humidity :  57.2057097606 %
```

Configuration seems to be correct (I have tried with 0x76 or 76 for address but the result is the same) : 

[image]

Using "root access" or not is not better : 

[image]

What ever I'm doing it's always the same result : 

[image]

other parts of this wonderful plugin are working well

Squirrelf, 2021-01-03:
> Hello evrerybody,
>
> I'm facing the same issue : my BME280 is working fine, it is correctly seen by the system :
>
> ```
> root@octoprint:/home/pi#  i2cdetect -y 1
>      0  1  2  3  4  5  6  7  8  9  a  b [...]

what worked for me was installing adafruit-circuitpython-bme280 with sudo (to be system wide as per the Adafruit instructions), in case you didn't do that. (haven't actually tried installing for current user only)
for the address 76
and then had to actually disable the default "use SUDO", and it "magically" worked. (with "use SUDO", I would get 0C 0%)

Will-It-Work-42, 2021-01-04:
Hi Squirrelf,

MANY thanks to you ! It works like a charm

I had to install pip3 before : 

`sudo apt install python3-pip`

And then as you said : 

`sudo pip3 install adafruit-circuitpython-bme280`

And disable "use SUDO"

and ... tada !

[image]

Hope this will help other users

f-hnt, 2021-01-04:
I have the same problem but with Si7021 sensor. Since the update in Sempember to Python3 I'm not able to use it as it shows zeros.
Unfortunately the last mentioned methods does not work for me. 

@vitormhenrique or somebody else can you suggest what could be wrong or alt least a workaround?

Thanks.

ainuke, 2021-01-24:
> what worked for me was installing adafruit-circuitpython-bme280 with sudo (to be system wide as per the Adafruit instructions), in case you didn't do that. (haven't actually tried installing for current user only)
> `sudo pip3 install adafruit-circuitpython-bme280`
> for the address 76 (didn't try 0x76)
> and then had to actually disable the default "use SUDO", and it "magically" worked. (with "use SUDO", I would get 0C 0%)
> 
> @vitormhenrique thanks for everything you have done 👍

I'm so psyched to try this. I've been considering downgrading Octoprint to Pv2.7 version just because of this problem with temp sensor. It drives my enclosure heater, which is the only way I can reliably print in winter (my basement is about 10-12C, and PLA warps big time on larger prints, sometimes detaching). I've been anxiously awaiting an update for Pv3 compatibility; @vitormhenrique is the greatest for keeping this plugin going, and I really appreciate all his dedication and hard work!

f-hnt, 2021-01-25:
Just to share my experience when I had readings "0" temp and "0" humidity. 

After doing all suggestions in this thread, for me the problem still persisted. 
For some other reasons I had to clear my cookies and cache from the browser (Firefox). After this clean-up the readings from the sensor now work.

Regards.

cerulean-yonder, 2021-02-04:
I've downloaded image 0.17.0 initially and updated as per above and got it working.
I had to do a reconfigure yesterday (re-image SD card) and downloaded the 0.18.0 version and tried this, but for me it isn't working now.
I've tried both Chrome and Firefox and clearing browser data, but unfortunately I'm still seeing "0" as the values. I've turned on logging for octoprint.plugins.enclosure but I can't see a log generated for this yet? I'm getting a sensor showing up correctly as per the "i2cdetect -y 1". I've also tried restarting using sudo enabled/disabled. No sure if I should downgrade to get functionality back or wait :)

geoffdavis, 2021-06-10:
I couldn't get the plugin to work even after installing the adafruit BME280 library as mentioned above.

However, it looks like GH-379 has a fix for this, and it works on my Octoprint 0.17 upgraded to python3 installation.

sirinsidiator, 2021-09-24:
I hit the same problem. After looking into it for a bit, I found that calling BME280.py directly gives me the following error:
```
pi@octopi:~/oprint/lib/python3.7/site-packages/octoprint_enclosure $ python BME280.py 0x77
Traceback (most recent call last):
  File "BME280.py", line 1, in <module>
    import smbus
ImportError: No module named smbus
```
Installing smbus via `sudo apt-get install python-smbus` fixed it for me and it now shows the values properly.

sajovicd, 2021-12-06:
Hello
Does anybody know why I have so many i2c values? I cant get BME280 to work...

Thank you

` i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
70: 70 71 72 73 74 75 76 77
`

Mcarbuhn, 2021-12-06:
I had the same problem we I wired the bme280 wrong.

sajovicd, 2021-12-07:
Hello,
Thnank you for fast respond... well I check the wires, check the wiring, check the pin, gpio... Have no more idea what culd be wrong. I have rpi3 b+ with the latest octopi installed.

JanHBade, 2022-02-23:
the problem still exists for me

only zeros are displayed :(

sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- 76 --                         
pi@Ender3Pro:~ $ ./bme280.py 

Temperature: 13.8 C
Humidity: 50.1 %
Pressure: 1011.2 hPa

VCOB, 2022-03-25:
I have a Pimoroni breakout bme280 connected to a hyper pixel 4.0 rectangular touch screen.  My i2c bus is i2c-22 so I modified instructions from the hyper pixel page to create a symbolic link for bus 1: sudo ln -s /dev/i2c-22 /dev/i2c-1.  Then I disabled the "Use SUDO" option as described above, but I also had to select "Use Board Pin #" as well.