vitormhenrique/OctoPrint-Enclosure #347: No Temp readings after plugin update

chimcen, 2020-09-30:
**Describe the bug**
After the last update of the enclosure plugin, octoprint reads no temp data anymore from my DHT22 connected on my Pi.
Before the last update it works like a charm.

**To Reproduce**
Steps to reproduce the behavior:
...

**Expected behavior**
Find a solution or help.

**Additional context**
No errors in octoprint.log.

robertomilan, 2020-10-05:
Same here. No errors in log.

chris948, 2020-10-11:
Is it just reading zero? Can you enable verbose logging? I had an issue yesterday where my DHT11 was reading 0*, but it tested fine with ./AdafruitDHT.py 11 4. There were obvious errors that in the .octoprint log that suggested disabling sudo. Doing that in the advanced section fixed it immediately.

vitormhenrique, 2020-10-12:
@chimcen have you tried disabling sudo? did it work? revert to python 2.7 if you are using 3.X because it won't work property on python 3.X for now...

chris948, 2020-10-12:
Also, and I have no historical knowledge, so this may be normal and expected, mine reads 0 until I actually start a print. Disabling sudo fixed it, but you have to be printing to confirm.

doudar, 2020-10-13:
Confirmed here also. I get the correct readings from the command line but the octoprint log says  "a bytes-like object is required, not 'str'" and recommends I try unchecking "sudo", which doesn't help in my case. 

Google suggests this is a change in the upgrade to python 3.5 .

doudar, 2020-10-14:
Works perfect after updating to @Yeraze fork:    https://github.com/Yeraze/OctoPrint-Enclosure/tree/Python_Stdout

Yeraze, 2020-10-14:
Happy to help @doudar .. I have a Pull Request in for you to merge whenever you feel like it @vitormhenrique .. It's a relatively minor and common thing in the 2.7->3.x conversion, everything in Python3 is Unicode strings or Byte arrays, so you have to convert them to utf-8 for most of the classical operations to work.

vitormhenrique, 2020-10-14:
Hi guys, I'm planing to completely re-write the plugin for Python 3.... 
I saw the pull request with that fix... I don't know if that is compatible with both python 3 and 2, so I'm afraid of just merging without testing.... and I don't have time to test it right now...

I worked on a PCB with PWM outputs, 5V level shifter for neopixels, 2 channel ADC and some more nice features that will be the base of development and extending this pluging in the future, I still need to solder the components... 

After that I will migrate all current libraries to python 3 with circuit python and actually installing the libraries with the plugin...

Next release will definitely be python 3 only, I'll not maintain python 2 as I hated to develop this on python 2.7 to begin with...

I'm gonna be posting more on this issue https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/302

And will close this one for now...

closed 2020-10-14 by vitormhenrique