vitormhenrique/OctoPrint-Enclosure #354: Not compatible on Pi 4

Arsenal10108, 2020-10-16:
It looks like the Adafruit_DHT library is deprecated and not compatible on the PI4 as the platform Detect doesn't know about the BCM 2711 chipset, thinks its not a PI. 

Adafuits site says that they support the CircuitPython Libraries, Which do work on the Pi4 is there a work around for this or a way to get the plugin to work on a PI4?

haplm, 2020-10-17:
I've got in working in shell (for python3) by following instruction here:
https://github.com/adafruit/Adafruit_Python_DHT

```pi@octopi:~/Adafruit_Python_DHT/examples $ ./AdafruitDHT.py 22 5
Temp=23.3*  Humidity=54.2%```

Unfortunately it doesn't work inside the plugin:
```2020-10-17 23:03:53,219 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2020-10-17 23:03:53,222 - octoprint.plugins.enclosure - WARNING - An exception of type TypeError occurred on log_error. Arguments:
("a bytes-like object is required, not 'str'",)
```

haplm, 2020-10-17:
I think I found it:
https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/456497f7e67067514e1711b0566db8988037897c/octoprint_enclosure/__init__.py#L880

It invokes python2, so this has to be changed to python3.

Then there's a problem with decoding the response.

```temp, hum = stdout.split("|")```

has to be changed to

```temp, hum = stdout.decode().split("|")```

I will create pull request tomorrow

haplm, 2020-10-18:
Actually, while my fix works for my specific environment, the situation is more complicated should this be deployed to others. The way how Enclosure plugin is written means that it is invoking python scripts via shell, instead of calling the code internally. And this is a problem with all the scripts.

Should this method be used going forward, it requires configurable python interpreter (2.7 vs. 3x), and perhaps even different handling of certain things (like .decode() needed before .split() on 3.x).

Try fixing your local installation in ```~/oprint/lib/python3.7/site-packages/octoprint_enclosure``` and let me know if you need some help. I will open a separate issue around this.

vitormhenrique, 2020-10-19:
this is all gonna be part of python 3 support as I'm ditching all those old libraries and changing the entire thing...

https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/302

closed 2020-10-19 by vitormhenrique