vitormhenrique/OctoPrint-Enclosure #355: Hardcoded version of Python for external scripts

haplm, 2020-10-18:
**Describe the bug**
Python version for external scripts (read DHT temperature, for example) is hardcoded as "python"
```cmd = sudo_str + "python " + script + str(sensor) + " " + str(pin)```

Since Octoprint is moving towards python3, I have installed Adafruit's libraries under python3, which means I need python3 as interpreter for the scripts.

Yes, I could setup alternatives and set python3 as python, but I'm not sure whether I'm not going to break something. It might be better to either make python executable configurable, or maybe stop calling parts of the plugin (in Python) as scripts?

Unfortunately there are also some parsing issues related to the version change, see #354 (title misleading, it works perfectly fine on RPi 4).

Could you suggest the best way moving forward?

**To Reproduce**
Steps to reproduce the behavior:
1. Install Adafruit's DHT libraries under Python3
2. Configure at least one DHT sensor
3. Check logs

**Additional context**
After the changes described in the linked issue (local edits), I can read data from DHT sensors just fine. I can help with some testing, but unfortunately I'm not a Python programer, so I don't dare to fix this for everybody.

vitormhenrique, 2020-10-20:
I'm also moving away from python2, all sensors and libraries will be installed on the same environment of the octoprint on the future version...  

A lot of people have issues installing and configuring libraries, so my plan is to include everything on the setup... and for external scripts I'll create a box input what python environment / file to use.....

You can follow python 3 thread here:

https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/302

domhardt, 2020-12-03:
It seems that I'm having the same problem. While I can successfully query the sensor values from the command line, the OctoPrint Plugin only shows 0°C and 0% RH for my AM2302. Please find the screenshots and log attached.

[image]

[image]

[image]

[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/5639028/octoprint.log)

vitormhenrique, 2020-12-03:
@domhardt you you are using python 3 this plugin is not yet compatible with it! I'm working on a major refactor but it takes time...

domhardt, 2020-12-04:
> @domhardt you you are using python 3 this plugin is not yet compatible with it! I'm working on a major refactor but it takes time...

Yes, I'm using the latest OctoPi version and Python 3. Keep on the good work!