vitormhenrique/OctoPrint-Enclosure #356: Time-based PWM output

haplm, 2020-10-18:
**Is your feature request related to a problem? Please describe.**
I'm using Waveshare DSI touchscreen on RPi, and the only way how to set brightness is to send it a PWM signal. Currently I'm using few cron entries to control it (like turn-off backlight at 8pm), but I would rather use Octoprint/Enclosure. 

**Describe the solution you'd like**
Current PWM functionality doesn't allow time-based output (with manual override). It would be really helpful to have a chance to set automated changes of brightness, as well as manual overrides within each interval.

What would be golden is support for a light intensity sensor on the input, plus time intervals :-)

**Describe alternatives you've considered**
One alternative is cron, which I use today (via ```gpio``` utility). Editing cron via terminal is not exactly something one wants to do, plus it doesn't allow for manual overrides.

I can imagine that combination of cron + basic PWM output in Enclosure could do the trick.

**Additional context**
```gpio``` doesn't support RPi 4 in default octopi image, it has to be upgraded. Plus people might not know how to work with cron. Having this feature would make time-based PWM tasks much easier to accomplish.

vitormhenrique, 2020-10-19:
when you say time based you mean like a schedule? for example form 8am to 10pm do this...  or do that ?

haplm, 2020-10-19:
If you want to keep it simple, then it would be enough to trigger specific action (send PWM value) at a configured time.

Basically I need to turn the display off at the evening, and turn it back on in the morning. Plus the manual override within the interval - ie. between the scheduled times I can set whatever value I like, but it get overwritten at the scheduled time.

vitormhenrique, 2020-10-19:
I want to double check how it's implemented right now...

and I' also completely re-writing the plugin to work o python 3 and be easier to maintain, it is a mess atm.... 

BUT if I'm not mistaken I only schedule and work on outputs "when" the printer starts a job...   as the idea of the plugin is to work when you are printing... 

for this to work how you want I would need to have a scheduler working with the server nonstop, and the overwrite could be confusion...  I'll think about how I could accomplish this and try to get something on the new version....

also next version I'm only planing to integrate PWM with an external i2c chip as I'm having way to many issues with RPI PWM..... just keep that in mind.... I'll start by implementing PCA9685 for PWM control...

haplm, 2020-10-20:
Thanks for the heads up with the PWM change. It actually works perfectly fine for me, but I have updated the gpio library which is outdated in the default Octopi installation and doesn't support 4B, I'm  driving the output using 74AHCT125, and using one of the pins that support hardware PWM.

PCA9685 is a cool board, but I can see a problem with that. Specs say "Adjustable frequency PWM up to about 1.6 KHz", while the crazy folks at Noctua implemented PWM which has acceptable range between 21kHz to 28kHz, so I'm afraid that all the people (like me) with their expensive Noctua fans will have a problem... (BTW I'm really not sure why Noctua did this, such precise control definitely looks like an overkill for a fan)

wmonzel, 2020-12-17:
Hi guys, 

@haplm: Just to be clear, it's not the "crazy folks at Noctua" that chose that frequency range.  It was the "crazy folks at Intel" :smile:, back in 2005.  I've only been able to find a dead link on Intel.com, but the paper does exist elsewhere.  Here: 
 https://www.glkinst.com/cables/cable_pics/4_Wire_PWM_Spec.pdf for example.  See page 10.  At a minimum, **any** PWM fan that adheres to the Intel spec will have to accept that frequency range.

haplm, 2020-12-18:
@wmonzel - yeah, you are right. I have realised that two weeks ago when I was building my first PC after like 20 years, and was mounting like 6 fans into the case :-).