vitormhenrique/OctoPrint-Enclosure #358: [Official Hardware] OctoPrint-Enclosure

vitormhenrique, 2020-10-20:
Hello Everyone, 

I started to work on the new version of the plugin and for that I'm creating an raspberry pi hat with some features to make it easy to deploy and maintain this plugin.

This will allow me test new stuff a little bit more easily and will probably use those shields on my enclosure...

It's basically a raspberry pi shield with the following features:

- PCA9635 PWM driver with 6 GPIO's broken out (control servos, led's and fans without need to use raspberry pi really poor pwm)
- ADS1015 12 bit ADC with two channels broken out, to be able to use thermistors as temperature inputs...
- Raspberry Pi fan connected to the PCA9635 driver (to be controlled based on rPI CPU temperature)
- One 5V level shifted output to control neopixels
- Bunch of extra GPIO outputs / input to be used, most of them with a convenient ground to be used as an input button, and couple with 5V and 3.3V that can be used with external relays
- I ordered the parts, now need to solder some PCB's and test the circuit...

All connectors will have JST-SH cables (other than servos and jumpers of course)

I'm interested to see if, and how many people would be interested on getting this board, so I can estimate a small batch and get it done....

Already had 5 ordered and manual soldered one to start testing and developing...

[image]
[image]

haplm, 2020-10-20:
I would love to have one, especially if there's a service that can ship an assembled one to Europe.

My only reservation is the fan. 30mm fans tend to be noisy. It might be better to use the space for additional electronics, and mount 40mm PWN controlled Noctua (or equivalent) on the side of the housing.

haplm, 2020-10-20:
Heh, look no further:
https://datasheets.maximintegrated.com/en/ds/MAX31790.pdf

sajovicd, 2020-10-20:
Count me in for one board.

gerthmar, 2020-10-20:
Nice work!! Are you planning to share the design sources?

I started something similar some time ago, but i prefer your idea to use a PCA9635 PWM Controller over my solution to use power FETs to control fans and leds. 

I'd like to share my thoughts about the shield I designed. Maybe you could use some of it for the 'official' hardware board.

How do you connect your 3d printer to the raspberry? Does the connection go thorugh the Hat? On my board, i added a JST connector to the Hat and built an adapter board to interface with my prusa i3 (https://www.thingiverse.com/thing:3953845).

I also added a JST connector to attach an I2C bus to the raspberry, which can interface with several MCP9808 temperature sensors digitally. I like the idea that i can place multiple sensors in the enclosure and just having a bus 3 wires to connect them.
For the MC9808, i also built a small PC which has low thermal masses and two connectors to daisy-chain multiple sensors. Unfortunately, I did not have the time yet to see if it works with the octoprint-enclosure plugin.

I also placed a 12V to 5V DC/DC converter to my board in order to power the raspi from the 12V, which i planned to use for other actors like fans or LED's. Looking at your design, I think i might have overdone the job here :-), since fans and leds don't neccesarily require 12V.

I placed two LED, one to 12V, one to 5V just to see if the voltages are present - i love LEDs to see whats going in, but thats just personal taste :)

Please let me know if I can assist you in any way with the design of the hardware. 

[image]
[image]

DrDrache, 2020-10-21:
I'd be in for a board - i've been thinking to go enclosed on my Printer

discreteaudio, 2020-10-30:
I'd be in for two boards

SimGas, 2020-11-10:
I'd also be in for a board (or could you share some design files preliminary?). I was actually searching quite a while for shields which do exactly what you describe here. I stopped searching a week ago and just stumbled over this today. I wonder why I did not find this thread earlier

I could also offer design support in Altium or Eagle

Three things I had on my list for a potential design:
- 5V input feed from the hat (Basically removes the need for the USB-powering). It would indeed bypass all power protection circuitry, but with a medium to high quality 5V Power supply this shouldn't make a difference. Could also be configured optionally with for example solder jumpers
- Smoke detector input: I would prefer a single one GPIO with screw terminals instead of a plugged connection for higher reliability.
- Regarding the fan, which has been mentioned before:
Most 30mm fans are quite noise. An option would be to either change to a 40mm footprint or, switch to a relatively quiet 30mm fan. I found that the Raspberry Pi Fan SHIM (minimal Fan shield) uses the AVC DATA0307R5H-002 fan which you can find on Aliexpress (I would personally prefer a 40mm fan, as sourcing the fan might be difficult if the supplier stops selling it)

haplm, 2020-11-10:
I very much agree with external 40mm fan (and save space on the board). The processor on 4B doesn't get hotter than 60º with no fan anyway. This is the enclosure that I have designed for mine, and which I can adapt to fit the board and possibly other components: https://www.prusaprinters.org/prints/41225-octoprint-box-raspberry-pi-4b-waveshare-43-dsi

I still think that it shouldn't be a hat, because separate board allows for better cooling and positioning inside an enclosure, but it seems it is just me :-).

@vitormhenrique - I have bought an oscilloscope, so I can eventually help to debug things

TazerReloaded, 2020-12-03:
Cool idea, if there is enough space, I'd use JST-XH through-hole connectors (like the BTT SKR 1.4 and many other boards), because they are way more common and pretty robust.
Unfortunately I won't need one in the near future, I made a few adapter boards when required from components I already had, so some of the MOSFETs may be Overkill to switch some LEDs, but they all work fine so far.
Here a list of what I built, maybe you can use some ideas:
- DHT22 adapter board: 4 pins to Raspi (3.3V, GND, 2x GPIO), 6 pins to 2 sensors (one for enclosure, one for filament storage. Two 6k8 pull-ups.
- High power output board: 2x L3803 Logic-Level MOSFETs with appropriate resistors, drives lighting and a laser module.
- Transistor switch board, just a little generic NPN with base resistor to switch the ATX PSU, protecting the Raspi GPIO.
- DS18B20 adapter board with 3k3 pull-up, I use this sensor to monitor cold-end temperature, in case the fan fails or something else goes very wrong.

I have a few other small boards, but they are not directly connected to the Pi, so I'll omit them here. If the printer has a fixed enclosure anyways, my modular approach works quite well and is easily expandable. For beginners, a plug and play shield would still be a great option though.

Jyde314, 2020-12-19:
I d be in for one board also (in Europe/France). Today I am using pi3b+ without fan. I ve crimping tool and prefer your flexible solution with JST connectors. Impressive job ! :-)

ChristophDerndorfer, 2021-01-01:
I'd also be interested in the board (I'm based in Europe as well).

jpregione, 2021-01-24:
I would be interested in a board and am in the U.S. Could one of the existing PI HATs like the Automation HAT be used with this plugin?

ZlatkoD, 2021-01-25:
Also interested in 1 board.
Europe based (Croatia).
Thank you for all the good work regarding enclosure control.

erikosvaldsson, 2021-01-26:
I would also be interested in a board, I'm in the EU!

MaxBec, 2021-01-28:
hey @vitormhenrique,

i just build my enclosure with 4 fans (2 for the electric bay) and two for the printer. I used the adafruit EMC2101 boards but would really like to test your board to save space.

truglodite, 2021-03-14:
It is a slick design, but unfortunately I'm already set with modular circuitry (with all the details and bell/whistles, my printer will remain the rats nest it is with or without a pi hat lol). OTOH, if I was building another rig from scratch your board has some key features that would be compelling to for a design around. I really like the hardware PWM, and level shifted io... and again you did an A+ job on the layout. I'm with haplm regarding the fan though... seems a bit on the bloated side for what this is, but that's just my opinion... being the type who would integrate this in a larger box containing many other wirey circuitry anyways, my opinion probably does not apply to what it could be used for. ;)

ac1416, 2021-04-09:
You have these available?  I am interested.

micheld93, 2021-07-13:
Really nice, I did something similar for my Neopixel ring but only for them.  
What is the status? Can we get this Octoshield manufactured from you?

jayzees, 2021-08-07:
count me in, another one for europe !!

bonjipoo, 2021-08-07:
Don't think this plugin is being maintained anymore.  I doubt the board will ever come to light.