vitormhenrique/OctoPrint-Enclosure #363: Feature Request: Auto Shutdown with delay (to cool down before turn off)

Mox2488, 2020-10-26:
I would like to be able to add a delay to the auto shutdown option.

I control the printer power supply via a relay via the enclosure plugin. Recently I started a print that ended in the middle of the night. The auto shutdown function is great. I thought about and would appreciate to switch off the printer-power only after a few minutes when the hotend has cooled down by the fan.

vitormhenrique, 2020-11-11:
the feature is already there, go on the settings and add a delay on the auto shutdown...

[image]

closed 2020-11-11 by vitormhenrique

rogery555, 2020-11-27:
The explanation of the shutdown delay was and is still a bit confusing to me.  First shouldn't the first part say when print STOPS instead of starts?   "Time delay in secconds to turn on GPIO when print **starts** "  Also of course seconds is spelled wrong. 

When I first read this description I was like what the heck does time zone have to do with a shutdown delay.   But after reading it several times it seems you are combining two completely different functions into one setting (Shutdown delay &  Specific shutdown time).  I think what makes it most confusing is the use of "when the print starts" and  "note that events will only be scheduled after a **print starts**"    I'm thinking NO I don't care when the print starts I just want this pin to turn off 5 minutes after the print STOPS!!!    So I think I need to put 300 in here to achieve this but not really sure lol