vitormhenrique/OctoPrint-Enclosure #364: Support for missing readings

Whats-name, 2020-10-26:
Hi, I'm a complete noob to Linux, Python and Pi and I'm looking for some help getting my DHT22 readings into the Enclosure plugin please as I cannot figure out what I am doing wrong after lots of hours of failing to fix it. Details:
- Pi B+ (yes, it is old) overclocked to 900Mhz running OctoPrint v1.4.2 with Enclosure plugin v4.13.1
- DHT 22 sensor on GPIO pin 4 and currently on 5v to see if it helped instead of 3.3v
- Running test script in terminal I get a response like this (must be run as sudo or fails to get readings):
[image]
- Running in sudo in the plugin give 0'c readings and throws an error as such in the Octoprint.log:
[image]
- I've read about this so have removed sudo in the advanced settings:
[image]

This leaves me with where I'm at now, no readings at all, not even a 0 reading:
[image]

I'm sure I've made a complete novice error but I cannot work it out, do you have any suggestions at all please?

vitormhenrique, 2020-10-26:
try the following:

`sudo rpi-update`

to make sure you are using a recent kernel, and
```
sudo apt-get update
sudo apt-get upgrade
```

to make sure you are using a recent RPi.GPIO.

You will also need to make sure you are a member of the gpio group:
``` 
sudo adduser pi gpio
sudo chown root.gpio /dev/gpiomem
sudo chmod g+rw /dev/gpiomem
sudo chown root.gpio /dev/mem
```

try running the script without sudo now...

vitormhenrique, 2020-10-26:
Also, after saving the settings try to reboot octoprint...

Whats-name, 2020-10-26:
Thank you for the very quick response, @vitormhenrique. 

I did these a few times on Saturday when installing the DHT 22:
`sudo apt-get update
sudo apt-get upgrade`

- user Pi says already has GPIO access when I type:
`sudo adduser pi gpio`

No feedback in Terminal at all when I type in these commands but I presume they are working as no errors are presented:
`sudo chown root.gpio /dev/gpiomem
sudo chmod g+rw /dev/gpiomem
sudo chown root.gpio /dev/mem`

Rebooted after all of the above and the test only works as sudo still so now I'm running...
`sudo rpi-update`
...then I'll try again!

Whats-name, 2020-10-26:
Ok, so I have run through every one of your steps in order now just to make sure (thank you for posting them). So RPi updated, then get update and upgrade, the other steps, the reboot. In Terminal sudo is still needed to get a result, and I'm guessing this is my fundamental issue that is stopping the plugin from retrieving values too?

[image]

Whats-name, 2020-10-26:
I'm there, issue resolved after far too many hours of fiddling. I went into the interfacing options on raspi-config (again) and re-did a few of them. I'm not sure which change made the difference (GPIO remote access was already on) but it worked. Thank you, @vitormhenrique for your help too.

[image]

fallonk, 2020-10-29:
I just decided to add an AM2302 temp sensor to my control box, and I'm having the same issue.  The enclosure plugin is only seeing 0' and 0% results.  I got the right sensor selected and the right GPIO entered as well.  All the prereq's are installed as well - and the example script works just fine (as both root (with sudo) and pi users).  So, I know the hardware and even the software actually works - but I can't get it to pull a reading in the plugin.  I've done some of the above poking around as well, and even ran rpi-update to update the firmware.  suggestions?

  | Temperature | Humidity
-- | -- | --
Internal Temperature | 0 °C | 0 %

```
------------------------------------------------------------------------------
OctoPrint version : 1.4.2
OctoPi version    : 0.17.0
------------------------------------------------------------------------------

pi@octopi:~/Adafruit_Python_DHT/examples $ ./AdafruitDHT.py 2302 4
Temp=29.4*  Humidity=33.9%
pi@octopi:~/Adafruit_Python_DHT/examples $ sudo !!
sudo ./AdafruitDHT.py 2302 4
[sudo] password for pi:
Temp=29.3*  Humidity=33.9%
pi@octopi:~/Adafruit_Python_DHT/examples $

```

vitormhenrique, 2020-11-11:
have you tried again disabling he root on the advanced section?

n0t49a1n, 2020-11-17:
> have you tried again disabling he root on the advanced section?

i have same issue and yes sudo is unticked

fallonk, 2020-12-10:
It does not matter if I sudo or not.  It looks like there's something odd about the pulling of the data into the variables.  I've even removed and reinstalled the plugin.  

`2020-12-10 00:10:02,791 - octoprint.plugins.enclosure - DEBUG - Temperature dht cmd: python /home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/getDHTTemp.py 2302 4
2020-12-10 00:10:04,007 - octoprint.plugins.enclosure - DEBUG - Dht result: b'26.9 | 31.4\n'
2020-12-10 00:10:04,008 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2020-12-10 00:10:04,008 - octoprint.plugins.enclosure - WARNING - An exception of type TypeError occurred on log_error. Arguments:
("a bytes-like object is required, not 'str'",)
2020-12-10 00:10:04,009 - octoprint.plugins.enclosure - DEBUG - Sensor AM2302 Temperature: 0.0 humidity 0.0`

pi@octopi:~ $ python -V
Python 2.7.16
pi@octopi:~ $ python /home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/getDHTTemp.py 2302 4
26.6 | 32.0

vitormhenrique, 2020-12-10:
@fallonk see how your octoprint is running under python3.7?  look at it's path...

To get it working on python 3.7 you manually need to change some lines of code inside of the plugin, as described on other issues....

II'm working on python 3.7 version of the plugin but it will take few moths