vitormhenrique/OctoPrint-Enclosure #366: Issues when controlling Enclosure from the iPhone app octopod

OCOLLET91, 2020-11-07:
**Describe the bug**
Control of the gpis outputs with octopod on my iPhone is erratic. I have to press several time the button to switch either on or off the led commanded by a relay on gpio
On the octoprint interface it works perfcetly...

**To Reproduce**
Steps to reproduce the behavior:
1. open the iPhone app octopod
2. want to switch off the led... press the led off button on octopod interface... does,'t work...
3, erratically ends to work...
4, same for less on...
5. the same functions works perfectly from octoprint interface. The issue is juste with octopod

**Expected behavior**
I just expect to have a reliable behavior just as with octoprint interface...

**Additional context**
Octoprint, enclosure plugin and octopod are all at last versions

Thanks for support!

vitormhenrique, 2020-11-11:
can you reproduce the issue on the localhost? connected on the  web browser?

if not, I' sorry but there is no much I can help you with, as I don't use octopod...

closed 2020-11-11 by vitormhenrique

OCOLLET91, 2020-11-11:
Hi

The issue appears only on my IPhone on the octopod app. It is really
erratic the symbol on/off appears erratically on or off... and the
switching requires several attempts to get successful... with no apparent
reason. I am thinking of kind of a latency in the transmission on the
network?

When on my computer on the Octoprint webpage / enclosure page it works
perfectly....

Any idea welcome for investigation !

Kind regards

Olivier

OCOLLET91, 2020-11-11:
For exemple off erratic displays in octopod :

Display shows LEDs and printer ON when in reality LEDs are OFF and printer
is ON:

Or it could shows LEDs and Printer OFF when LEDs are OFF and printer is
ON...
To switch ON the LEDs I found that it is necessary to have first the status
LED OFF in the display...

Strange erratic behavior...?... don’t know if I am clear?

Kind regards

Olivier

sajovicd, 2020-11-11:
Hay,
I belive that this is app related problem, not enclosure plugin.
I try it on my iphone and it doesnt work any of enclousure button.
So I suggest you that you ask octopod owner for help.

OCOLLET91, 2020-11-11:
OK thank you. I will ask to Octopod creator.

By the way which app do you use to control Enclosure from you smartphone ?

Kind regards

Olivier

sajovicd, 2020-11-11:
I use octoclient.
Anyway I just need to look every now and then if everything is ok