vitormhenrique/OctoPrint-Enclosure #368: DHT11 stopped working (fix suggested)

matthewfallshaw, 2020-11-16:
**Describe the bug**
My DHT11 was working… then it wasn't. I note that https://github.com/adafruit/Adafruit_Python_DHT is now marked as deprecated, and they point to CircuitPython as the successor. I hacked together a fix, which is documented below.

**A start for a fix**
This is hacked together, but pretty close to what's going to be required as the successor (handling the DHT22 & AM2302 previously handled by [getDHTTemp.py](https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/octoprint_enclosure/getDHTTemp.py)).

``` bash
sudo apt update
sudo apt -y upgrade
sudo apt -y install python3 python3-pip python-smbus i2c-tools libi2c-dev
sudo pip3 install --upgrade setuptools
sudo update-alternatives --install /usr/bin/python python /usr/bin/python2.7 1
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3.7 2

sudo raspi-config
# Interface options > I2C > Enable
# Interface options > SPI > Enable

sudo reboot

# test for i2c errors
sudo i2cdetect -y 1
pip3 install RPI.GPIO adafruit-blinka
```

### CircuitPython-DHT

``` bash
pip3 install adafruit-circuitpython-dht
sudo apt -y install libgpiod2
```

### Replace `oprint/lib/python2.7/site-packages/octoprint_enclosure/getDHTTemp.py`

``` python
import time
import board
import adafruit_dht

# Initial the dht device, with data pin connected to:
dhtDevice = adafruit_dht.DHT11(board.D23)

# you can pass DHT22 use_pulseio=False if you wouldn't like to use pulseio.
# This may be necessary on a Linux single board computer like the Raspberry Pi,
# but it will not work in CircuitPython.
# dhtDevice = adafruit_dht.DHT22(board.D18, use_pulseio=False)

try:
    # Print the values to the serial port
    temperature_c = dhtDevice.temperature
    humidity = dhtDevice.humidity
    print("{:.1f} | {} ".format(temperature_c, humidity))

except RuntimeError as error:
    # Errors happen fairly often, DHT's are hard to read, just keep going
    # print(error.args[0])
    print('-1 | -1')
except Exception as error:
    raise error
```

oeijoenj, 2020-11-16:
Is it working now with this fix ?

matthewfallshaw, 2020-11-16:
> Is it working now with this fix ?

My fix is working for me.
(I've updated my comment. See the note I added — my fix works only for DHT11, since I don't have DHT22 or AM2302 to test, and I didn't duplicate [the previous handling of those devices](https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/ff9d899d10c9bb9623d1c4b4eb16be1bc82ffdfc/octoprint_enclosure/getDHTTemp.py#L6-L8). This is totally not the "responsible contribution to an open source project I use and love" I'd like to make if I had more time; it's instead a completely half-done "ohh I hacked together a totally back yard patch that got me working, and someone [more responsible than me | with more time] might find useful as a start to doing this properly"… which is why I filed it as an Issue rather than a Pull request.)

oeijoenj, 2020-11-17:
Follow your instructions but seems still does not work n my Octopi. Still show 0 degress and 0% humidity.
I use pin 4 for the DHT11.
Did I miss something ? Thanks for your help

[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/5550764/octoprint.log)

oeijoenj, 2020-11-17:
You're awesome
[image]

oeijoenj, 2020-11-17:
btw, is it possible to add second sensor ? I saw that you put "dhtDevice = adafruit_dht.DHT11(board.D23)" in the script means we cannot change GPIO pin from the plugin. Is there a way so to change the script so we can change port from plugin setting ?
Thanks so much

n0t49a1n, 2020-11-17:
also the same error yet i cannot get this "fix" to work

meganssmith, 2020-11-21:
Got mine to work with my DT22, just had to change it to 

dhtDevice = adafruit_dht.DHT22(board.D4)
in oprint/lib/python2.7/site-packages/octoprint_enclosure/getDHTTemp.py

oeijoenj, 2020-11-21:
Just changed to DHT22 and work like charm too

Jugrnot, 2020-11-22:
> **Describe the bug**
> My DHT11 was working… then it wasn't. I note that https://github.com/adafruit/Adafruit_Python_DHT is now marked as deprecated, and they point to CircuitPython as the successor. I hacked together a fix, which is documented below and working for me.
> 
> **A start for a fix**
> This is hacked together, but pretty close to what's going to be required as the successor (handling the DHT22 & AM2302 previously handled by [getDHTTemp.py](https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/octoprint_enclosure/getDHTTemp.py)).
> 
> ```shell
> sudo apt update
> sudo apt -y upgrade
> sudo apt -y install python3 python3-pip python-smbus i2c-tools libi2c-dev
> sudo pip3 install --upgrade setuptools
> sudo update-alternatives --install /usr/bin/python python /usr/bin/python2.7 1
> sudo update-alternatives --install /usr/bin/python python /usr/bin/python3.7 2
> 
> sudo raspi-config
> # Interface options > I2C > Enable
> # Interface options > SPI > Enable
> 
> sudo reboot
> 
> # test for i2c errors
> sudo i2cdetect -y 1
> pip3 install RPI.GPIO adafruit-blinka
> ```
> 
> ### CircuitPython-DHT
> ```shell
> pip3 install adafruit-circuitpython-dht
> sudo apt -y install libgpiod2
> ```
> 
> ### Replace `oprint/lib/python2.7/site-packages/octoprint_enclosure/getDHTTemp.py`
> ```python
> import time
> import board
> import adafruit_dht
> 
> # Initial the dht device, with data pin connected to:
> dhtDevice = adafruit_dht.DHT11(board.D23)
> 
> # you can pass DHT22 use_pulseio=False if you wouldn't like to use pulseio.
> # This may be necessary on a Linux single board computer like the Raspberry Pi,
> # but it will not work in CircuitPython.
> # dhtDevice = adafruit_dht.DHT22(board.D18, use_pulseio=False)
> 
> try:
>     # Print the values to the serial port
>     temperature_c = dhtDevice.temperature
>     humidity = dhtDevice.humidity
>     print("{:.1f} | {} ".format(temperature_c, humidity))
> 
> except RuntimeError as error:
>     # Errors happen fairly often, DHT's are hard to read, just keep going
>     # print(error.args[0])
>     print('-1 | -1')
> except Exception as error:
>     dhtDevice.exit()
>     raise error
> ```
> 
> ### Note
> My fix only works for DHT11, where the old `getDHTTemp.py` worked (I assume from reading the code — I don't have the other devices to test them) for DHT11, DHT22 & AM2302.

So I went through this entire process (had to install libgpiod2 manually using dpkg) and when I reload the octoprint page, it displays the proper temp/humidity for about 2 seconds, then when the site is finished loading, it goes back to 32*/0% not working like before.

Any ideas?

kincaid-egotrip, 2020-11-22:
I had high hopes for this fix. Since I can get a reading from the script in terminal. 
My problem is that i run python3, maybe I just have to edit __init__.py from
cmd = sudo_str + "python " + script + str(sensor) + " " + str(pin)
to
cmd = sudo_str + "python3 " + script + str(sensor) + " " + str(pin)

I am mid print atm. so I cannot reboot octorpint to test. and have none result just editing the init file.

vitormhenrique, 2020-11-22:
This plugin does not support python 3 without some changes!

It’s all over other issues, I’m working on a new version for python 3 with
major code refactoring.

kincaid-egotrip, 2020-11-22:
> This plugin does not support python 3 without some changes! It’s all over other issues, I’m working on a new version for python 3 with major code refactoring.

Yeah, I know it doesn't support python3 yet, was only asking for a temporary fix. I tried changing __init__.py and had no success.I'm right on the doorstep of a solution for my issue. I've managed to get the values in cli, after a lot of hurdles. I'm running pi4 and also had issues with wiringpi not supporting my board layout. And a few other issues.

I still have hope to find a solutiontthat will work until you finish porting over to python3. I admire the people with skills to actuallymake tthis work.Me on the other hand has no knowledge about python language,and only tryingto reverse engineer on a noob level

vitormhenrique, 2020-11-22:
You would definitely need to change python to python3 if that is the python
that you installed the libraries on....
Also there is the issue with getting the variable from the terminal that
needs to be applied for python3...

kincaid-egotrip, 2020-11-22:
Changed to python3, and restarted octoprint service. but then octoprint gui would not load. Service was running, no error, so I had to reverse the changes to get it back up and running. I think I need to read up on python before I continue.

kincaid-egotrip, 2020-11-24:
I even tried to write a "hello world" test script with temp and humidity hardcoded to try to get an output.

I think I've figured out why it does not work for me. but not how to fix it. 
`2020-11-24 14:11:57,364 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.`
No wonder I don't get any data to the plugin, since the plugin can't even run getDHTtemp.py

I have sudo disabled, and I have tried enabling it, and get the same error.
Here is a caviat: I am running octoprint as root due to rpi4 spi dont work with rpi4 (go figure), so I have to run pwm and root.

kincaid-egotrip, 2020-11-24:
'2020-11-24 15:39:34,319 - octoprint.plugins.enclosure - DEBUG - Dht result: b'22.1 | 51.6\n''
'2020-11-24 15:39:44,382 - octoprint.plugins.enclosure - WARNING - An exception of type TypeError occurred on log_error. Arguments:'
'("a bytes-like object is required, not 'str'",)'

oeijoenj, 2020-11-24:
I have followed all procedure and using DHT22 and it works. I also put it in dashboard for easier to see. 

[image]

kincaid-egotrip, 2020-11-25:
i finally got it working:
edited init.py with:
`cmd = sudo_str + "python3 " + script + str(sensor) + " " + str(pin)`
`temp, hum = stdout.decode().split("|")`
finally results:
`2020-11-25 07:46:45,930 - octoprint.plugins.enclosure - DEBUG - Sensor Input 1 Temperature: 21.8 humidity 38.1`
temp and humidity now showing in plugin and on dashboard.

Wyzed17, 2020-11-26:
I could not get it working since I upgrade to python 3. I get the sensor working by console, but not by the plugin. I've tryed what was shown on this thread but not success. I did something wrong I guess, my best chance is to wait for the native compatibility with python 3 of the plugin.

This is my last log output:

2020-11-26 23:55:16,456 - octoprint.plugins.enclosure - WARNING - An exception of type NameError occurred on log_error. Arguments:
("name 'stdout' is not defined",)

I've done the process shown by the OP and this one later:

https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/368#issuecomment-733504482

P.D. N00bie question. In the OP, for the pin lane, (board.DXX) means the Board or BCM number? I get confused since I've seen in this post people leaving that as "board.D4"), but in the raspberry pinout, the "board pin n4" is a 5V Power pin. My sensor's attached to GPIO4 (Board pin 7). I guess that I should leave that as "board.D7"), should I?

kincaid-egotrip, 2020-11-27:
(board.DXX) is gpio pin number..
GPIO4 should then be (board.D4)

Your error seems to be different. Stdout variable is not defined.
Did you do changes to init.py or only getDHTtemp.py?

Wyzed17, 2020-11-27:
@kincaid-egotrip 

I did the change on both. Now I'm printing so I'm not sure if I can modify those on the way. However, here's two screenshots of my plugin config:

[image]
[image]

I noticed that I'm not able to change the BOARD or BCM reference for the GPIO pins. That checkbox sticks activated despite of the changes I'm appliying to it. Not a big deal, since I'm also using successfully a GPIO output for controling a LED strip, and works flawessly choosing the BOARD pin reference. I assume that it's the same for the DHT22. The sensor is attached to GPIO4, board pin number 7, isn't it? (Raspberry Pi 3B+). Anyways, I'm not getting any read in "4" (GPIO BCM), neither in "7" (GPIO BOARD). 

Another issue that I notice is that my path is slightly different from the seen on this post. I'm not sure what difference does it make: (.../python2.7/... -> .../python3.7/...)

oprint/lib/**python3.7**/site-packages/octoprint_enclosure/getDHTTemp.py
oprint/lib/**python3.7**/site-packages/octoprint_enclosure/__init__.py

Here's another few screenshots of both archives:
[image]

While I was writing this post I figured out that maybe my error came from change the "temp, hum" lane where I shouldn't. I change it from whatever it was writted to "**temp, hum = stdout.decode().split("|")**" in "**check_enclosure_temp**" and "**read_dht_temp**". I didn't change "**get_sensor_data**".

[image]
[image]
[image]

There's no doubt with the another lane, **cmd = sudo_str + "python3 " + script + str(sensor) + " " + str(pin)**, since I can only find it at one place on the file, located on "**read_dht_temp**", the last screenshot from above. I think that maybe I should only change lanes on the "**read_dht_temp**" function. If that's true, maybe somebody could provide me the original lane in "**check_enclosure_temp**" to avoid reinstall all the plugin?

As you can see, the sensor is working fine: 
[image]

I think that I gave all the needed information that I'm able to recognise. Sorry for this text brick UwU.
Cheers! Thanks in advance!

P.D. OctoPrint version 1.4.2
       OctoPi version 0.17.0

kincaid-egotrip, 2020-11-27:
@Wyzed17 
> 
> As you can see, the sensor is working fine:
> [image]
> 
> I think that I gave all the needed information that I'm able to recognise. Sorry for this text brick UwU.
> Cheers! Thanks in advance!
> 
> P.D. OctoPrint version 1.4.2
> OctoPi version 0.17.0

Try running getDHTtemp.py with python3 in CLI, and see what the output of that script is. Just to se if you get any output.
you shold see something like:
`21.3 | 39.6`

And I think you have done more changes in init.py than needed.
There are 2 lines under read_dht_temp

`cmd = sudo_str + "python " + script + str(sensor) + " " + str(pin)`
`temp, hum = stdout.split("|")`
change those two lines to:
`cmd = sudo_str + "python3 " + script + str(sensor) + " " + str(pin)`
`temp, hum = stdout.decode().split("|")`

The reason for the changes are, Running getDHTtemp.py as python3
and: since script is executed as python3, you need to decode the result back in, or else it tries to read bytes instead of string, (or was it the other way around?)
anywho: only 2 changes needed in init.py
 

The script OP has posted fixes depriciated libraries, 
my change, fixes python3 with new script.

matthewfallshaw, 2020-12-02:
I made some edits to the original post to clarify some of the things identified in the discussion above, and also…
… while I'm gratified that my terrible hack has been helpful to some folks, I do hope that someone has the time to turn this into a proper Pull Request that makes the obvious changes properly… and thereby permanently adds their name to (or adds another entry beside their name on) the rolls of the Beloved and Adored Contributors to Open Source Software. If you have the time, that could be *You*!

TazerReloaded, 2020-12-02:
Had the same Problem with my DHT22s, changing `temp, hum = stdout.split("|")` to `stdout.decode().split("|")` worked for me.
I run OctoPrint in a Python3 Venv, so `python` and `python3` point to the same binary.
By the way, installing the Adafruit plug-in with pip inside the Venv didn't work , installing from source did. So with these obstacles out of the way, my two DHT22 both work as expected, although sometimes reporting strange values (e.g. 3000% humidity). I guess that's caused by interference and not by software.

NA7KR, 2020-12-13:
The changes did not work for me still all 0.
from python shell looks good in interface 0
I did
cmd = sudo_str + "python " + script + str(sensor) + " " + str(pin)
temp, hum = stdout.split("|")
change those two lines to:
cmd = sudo_str + "python3 " + script + str(sensor) + " " + str(pin)
temp, hum = stdout.decode().split("|")

TazerReloaded, 2020-12-14:
Enable sensor debugging in settings and watch the OctoPrint log, should be some information what's going on there. The values are zero when an error happens while reading, and that should be printed to the log.

NA7KR, 2020-12-16:
This is what I get 
pi@octopi:~/OctoPrint/venv/lib/python3.7/site-packages/octoprint_enclosure $ python getDHTTemp.py 20
23.0 | 43

2020-12-16 19:29:05,573 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('not enough values to unpack (expected 2, got 1)',)

vitormhenrique, 2020-12-16:
@NA7KR you are using python 3, you need to apply this patch to get it working:

https://github.com/vitormhenrique/OctoPrint-Enclosure/pull/348/commits/16f3ec817877f94c18fc013c6e8a6bc4642ff7b3

NA7KR, 2020-12-16:
Still the same error

NA7KR, 2020-12-17:
seen you updated today and still the same error.

NA7KR, 2020-12-17:
fix it
`import time
import sys
import board
import adafruit_dht

if len(sys.argv) == 3:
    pin = sys.argv[2]
else:
    sys.exit(1)
# Initial the dht device, with data pin connected to:
dhtDevice = adafruit_dht.DHT11(pin)`

oeijoenj, 2020-12-26:
anyone use 2 sensor ? I want to put another sensor on filadryer if possible.
I added second sensor in plugin setting, it show some number but number remain showed even I take the sensor out. Seem it's not working then.

TazerReloaded, 2020-12-26:
I have 2 sensors working. Did you specify different pins for each sensor? Does it work if you call the script directly?

oeijoenj, 2020-12-26:
Haven't used script yet but I did specify other pin for 2nd sensor in plugin setting. 
Did you set pin in plugin setting or change it in script? 

> I have 2 sensors working. Did you specify different pins for each sensor? Does it work if you call the script directly?

TazerReloaded, 2020-12-26:
I specified everything in plugin settings. Only modification I made is described here: https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/368#issuecomment-737412289
My settings menu looks like this:
[image]

oeijoenj, 2020-12-26:
Thanks so much. 
Will give it a try again. 

> I specified everything in plugin settings. Only modification I made is described here: [#368 (comment)](https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/368#issuecomment-737412289)
> My settings menu looks like this:
> [image]

oeijoenj, 2020-12-26:
I added 2nd sensor in plugin setting and haven't put it on yet but temp dan humidity already showed in dashboard.
How did you this? changed stdout.split("|") to stdout.decode().split("|")

[image]

> Thanks so much.
> Will give it a try again.
> 
> > I specified everything in plugin settings. Only modification I made is described here: [#368 (comment)](https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/368#issuecomment-737412289)
> > My settings menu looks like this:
> > [image]

oeijoenj, 2020-12-27:
Hi.. could you post complete script for getDHTTemp.py ?
Currently I only can use 1 sensor and want to add another 1.

Thanks

> if len(sys.argv) == 3:
>     pin = sys.argv[2]
> else:
>     sys.exit(1)
> dhtDevice = adafruit_dht.DHT11(pin)

NA7KR, 2020-12-29:
full code 
`
#!/usr/bin/env python3
import time
import sys
import board
import adafruit_dht

if len(sys.argv) == 3:
    pin = sys.argv[2]
else:
    sys.exit(1)
dhtDevice = adafruit_dht.DHT11(pin)

# you can pass DHT22 use_pulseio=False if you wouldn't like to use pulseio.
# This may be necessary on a Linux single board computer like the Raspberry Pi,
# but it will not work in CircuitPython.
# dhtDevice = adafruit_dht.DHT22(board.D18, use_pulseio=False)

try:
    # Print the values to the serial port
    temperature_c = dhtDevice.temperature
    humidity = dhtDevice.humidity
    print("{:.1f} | {} ".format(temperature_c, humidity))

except RuntimeError as error:
    # Errors happen fairly often, DHT's are hard to read, just keep going
    # print(error.args[0])
    print('-1 | -1')
except Exception as error:
    dhtDevice.exit()
    raise error
`

oeijoenj, 2020-12-30:
Thanks so much.
I have edited getDHTTemp.py and seems still only 1 sensor works.
Is there any GPIO restriction? I use GPIO 5 and 26 but both did not work.
Also, since you said that you reinstall it, is that mean that only file getDHTTemp.py was changed ?
Did you change init.py as well ?

thanks 
 
> #!/usr/bin/env python3
>   import time
>   import sys
>   import board
>   import adafruit_dht
>   
>   
>   if len(sys.argv) == 3:
>       pin = sys.argv[2]
>   else:
>       sys.exit(1)
>   dhtDevice = adafruit_dht.DHT11(pin)
>  
>   try:
>       # Print the values to the serial port
>       temperature_c = dhtDevice.temperature
>       humidity = dhtDevice.humidity
>       print("{:.1f} | {} ".format(temperature_c, humidity))
>   
>   except RuntimeError as error:
>       # Errors happen fairly often, DHT's are hard to read, just keep going
>       # print(error.args[0])
>       print('-1 | -1')
>   except Exception as error:
>       dhtDevice.exit()
>       raise error

robertomilan, 2021-01-11:
AM2302 perfectly working with the fix for DHT22. THX!

Jugrnot, 2021-01-12:
I've one again combed through this entire thread and have tried every single suggestion mentioned, including wiping out the plugin entirely and reloading it then making changes again.

Thus far the only thing I have managed to accomplish is before tonight, while the octoprint page was loading, the sensor plugin showed the correct temp values..... now, while it loads, after it loads, ten years later............. it shows 0/0%.  Why is something that should be simple this difficult to accomplish????

vitormhenrique, 2021-01-12:
> I've one again combed through this entire thread and have tried every single suggestion mentioned, including wiping out the plugin entirely and reloading it then making changes again.
>
> Thus far the [...]

If you are using new releases of python 3, this plugin DOES NOT SUPPORT it, i said this many times, it is possible to make it work, but for more advanced users, the new release will support python 3 but it will take few months to be released!

matthewfallshaw, 2021-01-30:
Top comment updated with fixes suggested by others in this thread, and now working for me on python3 (with my super flakey DHT11, for values of "working" that return `-1 | -1` about four times in five… but that does appear to be enough to get a value).

I also updated the code to actually parse the command line arguments and choose DHT11 vs DHT22… which *might* make multiple sensors work?

(And I note @robertomilan commented that he had "AM2302 perfectly working with the fix for DHT22". I'm not sure what that means, but it might mean that if you use an AM2302 but configure it as a DHT22 it works?)

melrhombus, 2021-02-04:
Tried the steps in first post but getting an error outside enclosure plugin
python3.7 ./getDHTTemp.py 11 4

Traceback (most recent call last):
  File "./getDHTTemp.py", line 28, in <module>
    raise error
  File "./getDHTTemp.py", line 18, in <module>
    temperature_c = dhtDevice.temperature
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_dht.py", line 243, in temperature
    self.measure()
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_dht.py", line 190, in measure
    pulses = self._get_pulses_bitbang()
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_dht.py", line 133, in _get_pulses_bitbang
    with DigitalInOut(self._pin) as dhtpin:
  File "/home/pi/.local/lib/python3.7/site-packages/digitalio.py", line 118, in __init__
    self._pin = Pin(pin.id)
AttributeError: 'str' object has no attribute 'id'

hartmannf, 2021-03-17:
Hi 
Sorry to that question, but my understanding isn`t that well. I also tried to fix according your suggestions to get DHT22 Sensor working, but without success. in terminal it works. 
Did I understand it right, that you are working on new version that will take some time up to Adafruit Lib works properly on Oct enclosure plug in? 
Thanks for your great job.

YeastyCodpiece, 2021-03-27:
> My fix only works for DHT11 & DHT22, where the old `getDHTTemp.py` worked (I assume from reading the code — I don't have the other devices to test them) for DHT11, DHT22 & AM2302.

This fix worked for me (with a DHT22), thanks very much!

Felix-Jost, 2021-04-28:
> Hello
> Can somebody give me a hint on what am I doing wrong:
> 
> Code in simpletest2. *.py works, so wiring is ok.
> simpletest2.py looks like:
> `import time
> import board
> import adafruit_dht
> dhtDevice = adafruit_dht.DHT22(board.D4, use_pulseio=False)
> while True:
> try:
> temperature_c = dhtDevice.temperature
> humidity = dhtDevice.humidity
> print(
> "Temp: {:.1f} C Humidity: {}% ".format(
> temperature_c, humidity
> )
> )
> except RuntimeError as error:
> print(error.args[0])
> time.sleep(2.0)
> continue
> except Exception as error:
> dhtDevice.exit()
> raise error
> 
> ```
> time.sleep(2.0)
> ```
> 
> `
> 
> OctoPrint version : 1.5.3
> OctoPi version : 0.18.0
> 
> **pi@octopi:~ $ python3 simpletest2.py**
> Temp: 22..0)4 C Humidity: 37.8%
> Temp: 24.5 C Humidity: 37.4%
> Temp: 24.5 C Humidity: 37.4%
> Checksum did not validate. Try again.
> Temp: 24.5 C Humidity: 37.4%
> Temp: 24.5 C Humidity: 37.5%
> KeyboardInterrupt
> pi@octopi:~ $
> `
> 
> I tought to adapt the getDHTtemp.py script it might work. but it isn`t.
> #modif code : getDHTtemp.py looks like:
> 
> `import` time
> import board
> import sys
> import adafruit_dht
> 
> dhtDevice = adafruit_dht.DHT22(board.D4)
> dhtDevice = adafruit_dht.DHT22(board.D4, use_pulseio=False) #I also tried that version of code :(
> 
> try:
> temperature_c = dhtDevice.temperature
> humidity = dhtDevice.humidity
> print("{:.1f} | {} ".format(temperature_c, humidity))
> 
> except RuntimeError as error:
> 
> ```
> print('-1 | -1')
> ```
> 
> except Exception as error:
> dhtDevice.exit()
> raise error
> `
> 
> thanks a lot

did you have found the Issue ?

hartmannf, 2021-04-28:
> did you have found the Issue ?

Hello 

Let`s say it that way: No i didn`t find an Issue but a intermediate temporary turn arround until enclosure is raised up by vitormhenrique to python v3. 

I didn`t finished yet the coding but first looked promising: I out sourced all sensor and actors to an arduino and connected over I2C to the Raspberrypi where Octopi is running. See: Add i2c gpio support #390

Hope I could  help, and you get the temp into octopi. 

Here the arduino file but BE AWARE not finished and tested Yet!! 

`// Copyright: Raoul Hecky
//
// This library is free software; you can redistribute it and/or
// modify it under the terms of the GNU Lesser General Public
// License as published by the Free Software Foundation; either
// version 2.1 of the License, or (at your option) any later version.
//
//
//Libraries used:
// https://github.com/milesburton/Arduino-Temperature-Control-Library.git
// https://github.com/PaulStoffregen/OneWire

// extended and modified by Francis Hartmann

#include <Wire.h>
#include <DHT.h>
#include <DHT_U.h>

const byte I2C_ADDR = 0x08;

#define DHTTYPE DHT22

//In Out
const byte PIN_TEMP_HUM_ENCLOSURE_1 =2;
const byte PIN_TEMP_CTRL_ENCLOSURE_1PWM =3;
const byte PIN_TEMP_HUM_ENCLOSURE_2 =4;
const byte PIN_Temp_CTRL_ENCLOSURE_2=5;

const byte PIN_PRINTER =12;
const byte PIN_LIGHT =13;

// library 
DHT dht_ENC_1(PIN_TEMP_HUM_ENCLOSURE_1, DHTTYPE);
DHT dht_ENC_2(PIN_TEMP_HUM_ENCLOSURE_2, DHTTYPE);
float  temp_ENC_1=0,hum_ENC_1=0,temp_ENC_2=0,hum_ENC_2=0;

unsigned long lastTempRequest = 0;
int  delayInMillis = 0;

uint8_t lightState = HIGH;
uint8_t printerState = HIGH;

void setup()
{
    Serial.begin(115200);

    Wire.begin(I2C_ADDR);
    Wire.onReceive(receiveEvent);
    Wire.onRequest(requestEvent);

    dht_ENC_1.begin(); 
    dht_ENC_2.begin();
    
    pinMode(PIN_LIGHT, OUTPUT);
    digitalWrite(PIN_LIGHT, LOW);

    pinMode(PIN_PRINTER, OUTPUT);
    digitalWrite(PIN_PRINTER, LOW);

///    sensors.begin();
//    //disable blocking code
 //   sensors.setWaitForConversion(false);
 //   sensors.requestTemperatures();
 //   delayInMillis = 750; 
 //   lastTempRequest = millis();
}

void loop()
{
    BootUpDelay();
    requestTemperatureEnclosure_1();
    digitalWrite(PIN_LIGHT, lightState);
    digitalWrite(PIN_PRINTER, printerState);
}

void BootUpDelay()
{

}
void requestTemperatureEnclosure_1()
{
    if (millis() - lastTempRequest >= delayInMillis) // waited long enough??
    {

    temp_ENC_1 = dht_ENC_1.readTemperature();
    hum_ENC_1 = dht_ENC_1.readHumidity();    
    
    Serial.print("Temperature of Enclosure 1: ");
    Serial.print(temp_ENC_1);
    Serial.println("°C, ");
    Serial.print("Humidity of Enclosure 1: ");
    Serial.print(hum_ENC_1);
    Serial.println("%");
    delay(1000);

  
    lastTempRequest = millis(); 
    }
}

/* Action            | Operation | Register | Data to write
 * ------------------+-----------+----------+--------------
 * Set Light On      | Write     | 0x01     | 1
 * Set Light Off     | Write     | 0x01     | 0
 * Get Light state   | Read      | 0x01     |
 * Get Temp          | Read      | 0x02     |
 * Set Printer On    | Write     | 0x03     | 1
 * Set Printer Off   | Write     | 0x03     | 0
 * Get Printer state | Read      | 0x03     |
 */

const byte REG_LIGHT    = 0x01;
const byte REG_TEMP     = 0x02;
const byte REG_PRINTER  = 0x03;

uint8_t opcode; // register

void receiveEvent(int numBytes)
{
    //read register
    opcode = Wire.read();

    if (numBytes > 1)
    {
        if (opcode == REG_LIGHT)
        {
            uint8_t data = Wire.read();
            if (data == 0 || data == 1) //discard non valid values
                lightState = !data;
        }
        else if (opcode == REG_PRINTER)
        {
            uint8_t data = Wire.read();
            if (data == 0 || data == 1) //discard non valid values
                printerState = !data;
        }
    }
}

union floatBytes
{
    uint8_t bytes[4];
    float value; 
} floatContainer;

void requestEvent()
{
    if (opcode == REG_LIGHT)
    {
        Wire.write(!lightState);
    }
    else if (opcode == REG_PRINTER)
    {
        Wire.write(!printerState);
    }
    else if (opcode == REG_TEMP)
    {
        Serial.print("Read temp: ");
        Serial.print(temp_ENC_1);
        Serial.println("");
        
        floatContainer.value = temp_ENC_1;
        Wire.write(floatContainer.bytes, sizeof(float));
    }
}

`

akoirium, 2021-05-11:
Hello...
I have tried everything here....
And nothing works for me whit a dht22 :(

Hope it will be fix with the next release...

8465231, 2021-05-17:
I followed the OP but sensors still read 0 in octoprint? The sensor itself worked if I manually ran the example script on the main page, before doing all of this, so the sensor is hooked up properly. Now that example script no longer works though?

If the DHT22 sensor is not going to work, what sensor is recommended to use?

8465231, 2021-05-17:
Well, I needed to upgrade the SD card anyways, so took the time to reinstall everything from scratch and followed the OP exactly.

Yet it still just gives me a 0 reading on the sensor.

So any suggestions to get this working? Or better off moving to another sensor? What sensor is recommended?

akoirium, 2021-05-18:
Hello
The only sensor i succed to work is the DS18B20... see the readme...
But it's only for température 
All the other i tried don't works with the Newton raspiOS...

8465231, 2021-05-18:
Sad, I was wanting the humidity reading so I could keep track of the filament / humidity relationship in my garage. Temp is better then nothing though.

Any idea why the fix seems to work for most others but not for me? I know the sensor is working fine as manually getting the temp from it worked fine.

akoirium, 2021-05-18:
You have to know that humidity is not pertinent in the print chamber...
Because you get the humidity in air... not inside plastic...

For that i have made a DIY drybox / spooler with heating élément and a cheat température/humidity controler that i start 4 or 5 jours before printing for dry my filaments....

8465231, 2021-05-18:
Yeah, I am aware of that, more so that I can see what the ambient conditions are before/after printing to get an idea of how things react at different humidity levels. Not super important, just sucks to waste the money on these expensive sensors.

akoirium, 2021-05-18:
So wait for a release of octoprint-enclosure...
I hope me too... but there is the DS18B20 Who works great for this moment...

hartmannf, 2021-05-18:
> So wait for a release of octoprint-enclosure...
> I hope me too... but there is the DS18B20 Who works great for this moment...

Or connect an arduino over I2C and there you got DHT11/22 sensor working. There you can also hack other interesting things.  
By the way, I also didn`t got it running as it was originally planed, but that`s life. honestly I absolutely understand frustration :( but honestly victorhenrique did a great job with the enclosure plugin, and I am sure once he gots time and master done, there are going to be a working plug in.

8465231, 2021-05-18:
Oh, I am not blaming anyone or upset, just bummed lol.

Jugrnot, 2021-05-18:
> Or connect an arduino over I2C and there you got DHT11/22 sensor working. There you can also hack other interesting things.
> By the way, I also didn`t got it running as it was originally planed, but [...]

You have any quick documentation on this particular workaround?  It would be nice to have my octopi display the environmental sensors on my print enclosure again before the Raspberry Pi 9 is released.

[3 more replies]