vitormhenrique/OctoPrint-Enclosure #374: Configurable delay interval on GPIO inputs

numo68, 2020-11-28:
I am building a box with some added hardware where I want to start and stop the raspberry using a button. I want to use the same button to switch the hardware on and off; to protect against accidental shutdowns I would however like to require the user to hold it for at least a second.

There is a GPIO shutdown plugin where this can be parametrized, I however feel that this might be actually a common request in the scenarios the enclosure plugin is used for and maybe it would be nice to add the possibility to make the debounce interval configurable instead of the fixed one.

Many thanks

vitormhenrique, 2021-01-11:
could you send me a link for the plugin that you mentioned?

numo68, 2021-01-11:
https://plugins.octoprint.org/plugins/GPIOShutdown/